turtlebot/turtlebot4_simulator #10: Launch of multiple Turtlebotsv4 in one simulation

moutalibbadr, 2022-03-21:
**Launch multiple turtlebotsv4 without package conflicts**
Reconfigure the parameters, launch files, and configuration files of the turtlebotv4 to accommodate the launch of multiple robots in the same simulation.

**My attempt**
I have attempted to add namespace arguments in the nodes as well as in the launch files necessary to spawn one robot. My approach was to arrive at isolating one robot launch file that will call all the packages necessary to spawn a complete Turtlebot 4 in the simulation with a distinct namespace and tf_prefix. However this does not prove to be an easy task because of the multitude of nested launch calls and config files. So fare i haven't been able to successfully launch a turtlebot with a separate namespace.

roni-kreinin, 2022-03-21:
Yes there will be some challenges with having multiple robots running at the moment. We would need [create3_sim](https://github.com/iRobotEducation/create3_sim) to support multiple robots before the turtlebot4 sim can support it.

I won't be able to work on this until after the official launch, but if you want to work on a PR I can provide some assistance.

roni-kreinin, 2023-06-14:
Multiple robots are now supported in humble.

closed 2023-06-14 by roni-kreinin