turtlebot/turtlebot4_simulator #14: ACML issues

ahcorde, 2022-04-27:
**Please provide the following information:**
 - OS: Ubuntu 20.04
 - ROS Distro: e.g. Galactic
 - Built from source

 **Expected behaviour**

Run `amcl_core` without issues. This might be a nav2 issue.

 **Actual behaviour**

I'm getting this error:

```bash
[amcl-37] /opt/ros/galactic/lib/nav2_amcl/amcl: symbol lookup error: /opt/ros/galactic/lib/libamcl_core.so: undefined symbol: _ZN9nav2_amcl20LikelihoodFieldModelC1EddddmP5map_t
```

**To Reproduce**
Provide the steps to reproduce:

 ```
reset && ros2 launch turtlebot4_ignition_bringup ignition.launch.py slam:=sync rviz:=true localization:=true
```

**Other notes**

I compiled `nav2_acml` and I'm not getting this error. But I'm not able to set the initial position and navigate through the map. I can see this error:

```bash
[rviz2-15] [ERROR] [1651093133.026960494] [rviz2]: navigate_to_pose action server is not available. Is the initial pose set?
````

[image]

As you can see in this gif, I can't see the ACML particles when I set the initial pose. This means that we can't navigate too.

roni-kreinin, 2022-04-29:
Looking into it. I would recommend for now just using SLAM and nav2 together, without localization.

roni-kreinin, 2022-05-04:
Found the issue. The Nav2 AMCL package uses a `sensors_lib` library which is the same name as the sensors library built by `irobot_create_ignition_toolbox`. I am making a PR to change that library name.

roni-kreinin, 2022-05-04:
See https://github.com/iRobotEducation/create3_sim/pull/178

roni-kreinin, 2022-05-09:
Closing as the error has been fixed.

closed 2022-05-09 by roni-kreinin