turtlebot/turtlebot4_simulator #16: Humble Support

vinnnyr, 2022-05-26:
**Describe the the feature you would like**
Creating this ticket to discuss `humble` support, which would be nice to replace Turtlebot3 simulation in Gazebo with Turtlebot4 simulation in Ignition Fortress. 

**Other notes**
I am using the following Dockerfile to attempt to build in `Humble`

```
FROM ros:humble
RUN apt update && \
    apt install -qyy wget
RUN sudo sh -c 'echo "deb http://packages.osrfoundation.org/gazebo/ubuntu-stable `lsb_release -cs` main" > /etc/apt/sources.list.d/gazebo-stable.list' && \
    wget http://packages.osrfoundation.org/gazebo.key -O - | sudo apt-key add -
RUN apt update && \
    apt install -qyy ignition-fortress
WORKDIR /opt/tb4_ws/src/turtlebot4_simulator
COPY . .
WORKDIR /opt/tb4_ws/src
RUN vcs import < turtlebot4_simulator/dependencies.repos
WORKDIR /opt/tb4_ws
RUN rosdep install --from-paths src --ignore-src
RUN colcon build
```

vinnnyr, 2022-05-26:
I am currently thinking I need to see if we can get rosdep installations of the following dependencies for `humble` before I think it is possible for this package to be released in `humble` 

(when using the above Dockerfile)

```
irobot_create_ignition_bringup: Cannot locate rosdep definition for [ign_ros2_control]
irobot_create_ignition_plugins: No definition of [ignition-gui5] for OS version [jammy]
turtlebot4_ignition_gui_plugins: No definition of [ignition-gui5] for OS version [jammy]
turtlebot4_navigation: Cannot locate rosdep definition for [nav2_simple_commander]
```

roni-kreinin, 2022-05-26:
Humble support is planned but will take some time. We are looking to support Humble on the physical robot first, then add support for Humble/Fortress on the simulator. And yes, we will need [create3_sim](https://github.com/iRobotEducation/create3_sim) and other dependencies to support Humble first.

roni-kreinin, 2023-06-14:
Humble is fully released.

closed 2023-06-14 by roni-kreinin