turtlebot/turtlebot4_simulator #20: Rplidar messages are getting dropped

Srimathi-tw, 2022-09-19:
**Please provide the following information:**
 - OS:Ubuntu 20.04
 - ROS Distro:  Galactic
 - Built from source or installed: source
 - Package version: latest

 **Expected behaviour**
 On running 
  ``ros2 launch turtlebot4_ignition_bringup ignition.launch.py model:=lite rviz:=true slam:=sync``
and moving the bot around slam map to be generated

 **Actual behaviour**
` Message Filter dropping message: frame 'turtlebot4/rplidar_link/rplidar' at time 1.216 for reason 'discarding message because the queue is full'`
Getting the above error and as a result obstacles are not detected

**To Reproduce**

1. Run  ``ros2 launch turtlebot4_ignition_bringup ignition.launch.py model:=lite rviz:=true slam:=sync``
2.  Try driving the bot [https://turtlebot.github.io/turtlebot4-user-manual/tutorials/driving.html](https://turtlebot.github.io/turtlebot4-user-manual/tutorials/driving.html) by following the manual
Slam map will not be generated properly as obstacles are not detected

**Logs for reference**

`[planner_server-4] [INFO] [1663570054.867082422] [global_costmap.global_costmap_rclcpp_node]: Message Filter dropping message: frame 'turtlebot4/rplidar_link/rplidar' at time 1.184 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[sync_slam_toolbox_node-2] [INFO] [1663570054.902871461] [slam_toolbox]: Message Filter dropping message: frame 'turtlebot4/rplidar_link/rplidar' at time 1.216 for reason 'discarding message because the queue is full'`

Srimathi-tw, 2022-09-19:
@roni-kreinin  
1) We are running in quadro t1000 graphics card machine. Is there any minimum requirement on the same?
2) We are currently using ignition gazebo edifice version, could you please tell us which version you are using?

roni-kreinin, 2022-09-19:
There aren't any specific minimum requirements but generally a PC with an i5 and dedicated GPU should work fine.

Also I am using Edifice as well. Fortress support will be coming with the Humble update.

Srimathi-tw, 2022-09-20:
We tried building from binary still facing the same issue. Can you please share your rplidar properties @roni-kreinin

roni-kreinin, 2022-09-21:
I am using the default RPLIDAR [description](https://github.com/turtlebot/turtlebot4/blob/galactic/turtlebot4_description/urdf/sensors/rplidar.urdf.xacro). 

As for `irobot_create_msgs`, you will likely want to use the 1.2.4 tag, or the galactic branch. 2.0.0 and onwards is the Humble build, and the main branch also targets Humble.

Can you try running the simulation without SLAM and check in RVIZ that the lidar data looks correct?
i.e. `ros2 launch turtlebot4_ignition_bringup ignition.launch.py model:=lite rviz:=true`

If it does not look correct, do you have another PC you could test the simulation on? Or can you try to install the simulation on a VM?

Srimathi-tw, 2022-09-22:
@roni-kreinin We tried in 3 different systems one with dedicated graphics card and other 2 without but the issue persisted in all system. We also tried running via docker but it was the same case

roni-kreinin, 2022-09-22:
Can you try running the standard turtlebot4 and see if you get the same issue?

Srimathi-tw, 2022-09-22:
@roni-kreinin yes we tried and the issue persisted

roni-kreinin, 2022-09-22:
I have tested this again in a VM and still cannot reproduce your issue. 

[image]

Can you detail the installation steps that you took?

Srimathi-tw, 2022-09-22:
**Installation steps we followed while trying installation via binary** 
1. Installed ros2 galactic by following [https://docs.ros.org/en/galactic/Installation/Ubuntu-Install-Debians.html#additional-rmw-implementations-optional](https://docs.ros.org/en/galactic/Installation/Ubuntu-Install-Debians.html#additional-rmw-implementations-optional)
2. Dev tools
```
sudo apt install -y \
python3-colcon-common-extensions \
python3-rosdep \
python3-vcstool
```
3. Installed edifice via 

```
sudo apt-get update && sudo apt-get install wget
sudo sh -c 'echo "deb http://packages.osrfoundation.org/gazebo/ubuntu-stable `lsb_release -cs` main" > /etc/apt/sources.list.d/gazebo-stable.list'
wget http://packages.osrfoundation.org/gazebo.key -O - | sudo apt-key add -
sudo apt-get update && sudo apt-get install ignition-edifice
```
4. Debian installation 
```
sudo apt update
sudo apt install ros-galactic-turtlebot4-simulator ros-galactic-irobot-create-nodes
```

2 to 4 reference [https://turtlebot.github.io/turtlebot4-user-manual/software/turtlebot4_packages.html#installation-3](https://turtlebot.github.io/turtlebot4-user-manual/software/turtlebot4_packages.html#installation-3)

@roni-kreinin for your reference

Srimathi-tw, 2022-09-29:
@roni-kreinin were you able to reproduce the issue? . After I changed the values to 
```
  <gazebo reference="${name}_link">
    <xacro:ray_sensor sensor_name="${name}" gazebo="${gazebo}" 
                  update_rate="62.0" visualize="1" 
                  h_samples="640" h_res="1.0" h_min_angle="${-pi}" h_max_angle="${pi}" 
                  r_min="0.164" r_max="12.0" r_res="0.01"  v_samples="16" v_res="1.0" v_min_angle="-0.001" v_max_angle="-0.001" >
                  <plugin name="dummy" filename="dummyfile"></plugin>
    </xacro:ray_sensor>
    <xacro:material_darkgray/>
  </gazebo>
```
in `rplidar.urdf.xacro` it is working fine as a 2d lidar

roni-kreinin, 2022-09-29:
I tested this in a VM too and was not able to reproduce the issue there. The default vertical values are `v_samples:=1 v_res:=1 v_min_angle:=0 v_max_angle:=0` so its strange that your change fixed it.

ncnynl, 2022-11-01:
I tested turtlebot4 to build a map, and I also have this problem. I cannot get the activation data, and the laser line is not displayed on the ignition interface. Does this definitely require the GPU to obtain the data correctly, if the CPU cannot be used? @roni-kreinin

ncnynl, 2022-11-01:
I installed turtlebot4 simulator on nuc, there is no GPU support, the installation is all right, but open ros2 launch turtlebot4_ignition_bringup ignition.launch.py model:=lite rviz:=true The default scan topic output data is 0, also on ingition There is no laser like the one in your picture. @roni-kreinin

roni-kreinin, 2022-11-01:
@ncnynl Ignition Edifice only supports GPU rays and not CPU rays. As a result, a GPU is required to properly render the lidar and IR sensors. I am hoping that CPU rays will be supported in Ignition Fortress as we update to Humble/Fortress for TTB4.

ncnynl, 2022-11-04:
@roni-kreinin  I found your previous reply, https://github.com/iRobotEducation/create3_sim/blob/main/irobot_create_common/irobot_create_description/urdf/create3.urdf.xacro#:~:text=%3Crender_engine%3Eogre%3C/render_engine% 3E
I found that if ogre2 is used in the model, both the camera and the radar can obtain data normally, and the radar data can be obtained even if the CPU is used

roni-kreinin, 2022-11-04:
@ncnynl Are you able to get decent performance with ogre2? From my testing I would get significantly worse performance with the camera on ogre2, which is why I set it to ogre.

ncnynl, 2022-11-05:
At present, we can't use ogre normally, only ogre2 can, but there is no obvious difference in performance. At present, the speed of simulation movement is relatively slow, is there any way to change the speed, or it will take time to complete a map or navigation It is relatively long. The speed is the same whether it is orge or ogre2. It may be related to the speed limit of turtlebot4 itself. I want to adjust it to 2 m/s under simulation, so that it will be more convenient to test.  @roni-kreinin

ncnynl, 2022-11-05:
At present, I want to install turtlebot4_simulator on another nuc and virtual machine. I found that the same installation process, using orge2 will not work, and there will be a crash. @roni-kreinin 

```
[ign gazebo-1] ign gazebo server: /var/lib/jenkins/workspace/ogre-2.1-debbuilder/repo/RenderSystems/GL3Plus/src/GLSL/OgreGLSLProgramManager.cpp:623: void Ogre::GLSLProgramManager::extractUniformsFromProgram(GLuint, const GpuConstantDefinitionMap*, const GpuConstantDefinitionMap*, const GpuConstantDefinitionMap*, const GpuConstantDefinitionMap*, const GpuConstantDefinitionMap*, const GpuConstantDefinitionMap*, Ogre::GLUniformReferenceList&, Ogre::GLAtomicCounterReferenceList&, Ogre::GLUniformBufferList&, Ogre::SharedParamsBufferMap&, Ogre::GLCounterBufferList&): Assertion `size_t (arraySize) == newGLUniformReference.mConstantDef->arraySize && "GL doesn't agree with our array size!"' failed.
[ign gazebo-1] Stack trace (most recent call last) in thread 11231:
[ign gazebo-1] #31   Object "[0xffffffffffffffff]", at 0xffffffffffffffff, in 
[ign gazebo-1] #30   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7faddcb6c132, in clone
[ign gazebo-1] #29   Object "/lib/x86_64-linux-gnu/libpthread.so.0", at 0x7faddca30608, in 
[ign gazebo-1] #28   Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7fadd88436b3, in 
[ign gazebo-1] #27   Object "/usr/lib/x86_64-linux-gnu/ign-gazebo-5/plugins/libignition-gazebo-sensors-system.so", at 0x7fada84c9d87, in ignition::gazebo::v5::systems::SensorsPrivate::RenderThread()
[ign gazebo-1] #26   Object "/usr/lib/x86_64-linux-gnu/ign-gazebo-5/plugins/libignition-gazebo-sensors-system.so", at 0x7fada84c976e, in ignition::gazebo::v5::systems::SensorsPrivate::RunOnce()
[ign gazebo-1] #25   Object "/lib/x86_64-linux-gnu/libignition-sensors5.so.5", at 0x7fada8271c2e, in ignition::sensors::v5::Manager::RunOnce(std::chrono::duration<long, std::ratio<1l, 1000000000l> > const&, bool)
[ign gazebo-1] #24   Object "/lib/x86_64-linux-gnu/libignition-sensors5.so.5", at 0x7fada8286ebd, in ignition::sensors::v5::Sensor::Update(std::chrono::duration<long, std::ratio<1l, 1000000000l> > const&, bool)
[ign gazebo-1] #23   Object "/usr/lib/x86_64-linux-gnu/libignition-sensors5-rgbd_camera.so", at 0x7fad29b2cc05, in ignition::sensors::v5::RgbdCameraSensor::Update(std::chrono::duration<long, std::ratio<1l, 1000000000l> > const&)
[ign gazebo-1] #22   Object "/lib/x86_64-linux-gnu/libignition-sensors5-rendering.so.5", at 0x7fada82b0845, in ignition::sensors::v5::RenderingSensor::Render()
[ign gazebo-1] #21   Object "/usr/lib/x86_64-linux-gnu/ign-rendering-5/engine-plugins/libignition-rendering-ogre2.so", at 0x7fad876af8ff, in ignition::rendering::v5::Ogre2DepthCamera::Render()
[ign gazebo-1] #20   Object "/lib/x86_64-linux-gnu/libOgreMain.so.2.1.0", at 0x7fad8620d74b, in Ogre::Root::renderOneFrame()
[ign gazebo-1] #19   Object "/lib/x86_64-linux-gnu/libOgreMain.so.2.1.0", at 0x7fad8620d5d8, in Ogre::Root::_updateAllRenderTargets()
[ign gazebo-1] #18   Object "/lib/x86_64-linux-gnu/libOgreMain.so.2.1.0", at 0x7fad8631c6ad, in Ogre::CompositorManager2::_updateImplementation(Ogre::SceneManagerEnumerator&, Ogre::HlmsManager*)
[ign gazebo-1] #17   Object "/lib/x86_64-linux-gnu/libOgreMain.so.2.1.0", at 0x7fad863376e3, in Ogre::CompositorWorkspace::_update()
[ign gazebo-1] #16   Object "/lib/x86_64-linux-gnu/libOgreMain.so.2.1.0", at 0x7fad8632221e, in Ogre::CompositorNode::_update(Ogre::Camera const*, Ogre::SceneManager*)
[ign gazebo-1] #15   Object "/lib/x86_64-linux-gnu/libOgreMain.so.2.1.0", at 0x7fad8634a9c7, in Ogre::CompositorPassScene::execute(Ogre::Camera const*)
[ign gazebo-1] #14   Object "/lib/x86_64-linux-gnu/libOgreMain.so.2.1.0", at 0x7fad861ed57e, in Ogre::RenderTarget::_updateViewportRenderPhase02(Ogre::Viewport*, Ogre::Camera*, Ogre::Camera const*, unsigned char, unsigned char, bool)
[ign gazebo-1] #13   Object "/lib/x86_64-linux-gnu/libOgreMain.so.2.1.0", at 0x7fad860371e9, in Ogre::Camera::_renderScenePhase02(Ogre::Camera const*, Ogre::Viewport*, unsigned char, unsigned char, bool)
[ign gazebo-1] #12   Object "/lib/x86_64-linux-gnu/libOgreMain.so.2.1.0", at 0x7fad8621b4ae, in Ogre::SceneManager::_renderPhase02(Ogre::Camera*, Ogre::Camera const*, Ogre::Viewport*, unsigned char, unsigned char, bool)
[ign gazebo-1] #11   Object "/lib/x86_64-linux-gnu/libOgreMain.so.2.1.0", at 0x7fad861d3848, in Ogre::RenderQueue::render(Ogre::RenderSystem*, unsigned char, unsigned char, bool, bool)
[ign gazebo-1] #10   Object "/lib/x86_64-linux-gnu/libOgreMain.so.2.1.0", at 0x7fad861d13d5, in Ogre::RenderQueue::renderGL3(bool, bool, Ogre::HlmsCache*, Ogre::RenderQueue::RenderQueueGroup const&, Ogre::IndirectBufferPacked*, unsigned char*, unsigned char*)
[ign gazebo-1] #9    Object "/lib/x86_64-linux-gnu/libOgreHlmsPbs.so.2.1.0", at 0x7fada80c5f92, in Ogre::HlmsPbs::createShaderCacheEntry(unsigned int, Ogre::HlmsCache const&, unsigned int, Ogre::QueuedRenderable const&)
[ign gazebo-1] #8    Object "/usr/lib/x86_64-linux-gnu/OGRE-2.1/OGRE/RenderSystem_GL3Plus.so", at 0x7fad8466f17c, in Ogre::GL3PlusRenderSystem::_setPipelineStateObject(Ogre::HlmsPso const*)
[ign gazebo-1] #7    Object "/usr/lib/x86_64-linux-gnu/OGRE-2.1/OGRE/RenderSystem_GL3Plus.so", at 0x7fad84628cb0, in Ogre::GLSLMonolithicProgramManager::getActiveMonolithicProgram()
[ign gazebo-1] #6    Object "/usr/lib/x86_64-linux-gnu/OGRE-2.1/OGRE/RenderSystem_GL3Plus.so", at 0x7fad84626e3f, in Ogre::GLSLMonolithicProgram::activate()
[... cut ...]
[ign gazebo-1] #2    Object "/lib/x86_64-linux-gnu/libQt5Qml.so.5", at 0x7fadd59e2ddc, in QQmlPropertyCache::~QQmlPropertyCache()
[ign gazebo-1] #1    Object "/lib/x86_64-linux-gnu/libQt5Qml.so.5", at 0x7fadd59e28e8, in QQmlPropertyCache::~QQmlPropertyCache()
[ign gazebo-1] #0    Object "/lib/x86_64-linux-gnu/libQt5Qml.so.5", at 0x7fadd57bae30, in 
[ign gazebo-1] Segmentation fault (Address not mapped to object [0x7fad909c7460])
[INFO] [ign gazebo-1]: process has finished cleanly [pid 10701]

```

hilary-luo, 2023-09-13:
This issue is being closed due to inactivity. If you are still experiencing the issue, feel free to reopen this ticket when you are ready to continue the troubleshooting process.

closed 2023-09-13 by hilary-luo

HyydenJakcz, 2023-10-12:
> @roni-kreinin I found your previous reply, https://github.com/iRobotEducation/create3_sim/blob/main/irobot_create_common/irobot_create_description/urdf/create3.urdf.xacro#:~:text=%3Crender_engine%3Eogre%3C/render_engine% 3E I found that if ogre2 is used in the model, both the camera and the radar can obtain data normally, and the radar data can be obtained even if the CPU is used
> 
> The effect is like this tutorial I wrote https://www.ncnynl.com/archives/202210/5587.html

@ncnynl thank you for your solution. I tried it as a last ditch effort and somehow worked to fix it. However, like @roni-kreinin said it is visibly affecting the performance of my simulation

hilary-luo, 2023-10-12:
@HyydenJakcz if you have the freedom to, try upgrading to Humble, there have been a number of improvements made for the Humble version of the packages.

HyydenJakcz, 2023-10-13:
My setup at the moment is already using Humble. However, I tried what @ncnynl mentioned as a last-ditch effort when I saw that it was not really distro-dependent.

Flamethr0wer, 2024-07-03:
So the solution is to use Ogre2? What if I'm on a Raspberry Pi? Its GPU doesn't support OpenGL3.3 and Ogre2 demands it...

hylreg, 2024-10-19:
> 我找到了你之前的回复，https://github.com/iRobotEducation/create3_sim/blob/main/irobot_create_common/irobot_create_description/urdf/create3.urdf.xacro#:~:text=%3Crender_engine%3Eogre%3C/render_engine% 3E 我发现如果模型中使用了 ogre2，摄像头和雷达都可以正常获取数据，即使使用 CPU 也可以获取雷达数据
> 
> 效果就像我写的这个教程 https://www.ncnynl.com/archives/202210/5587.html

> 我找到了你之前的回复，https://github.com/iRobotEducation/create3_sim/blob/main/irobot_create_common/irobot_create_description/urdf/create3.urdf.xacro#:~:text=%3Crender_engine%3Eogre%3C/render_engine% 3E 我发现如果模型中使用了 ogre2，摄像头和雷达都可以正常获取数据，即使使用 CPU 也可以获取雷达数据
> 
> 效果就像我写的这个教程 https://www.ncnynl.com/archives/202210/5587.html

ogre不是默认使用GPU吗，我电脑上是有GPU的，但是收不到雷达数据。使用ogre2后，使用cpu运算，能收到雷达数据。想问一下ogre是不是不支持在仿真中使用GPU，只能在真机上使用GPU？

AS-HARI, 2025-04-21:
Hi @roni-kreinin.

OS:Ubuntu 22.04
ROS Distro: Humble
Built from source or installed: source ( git clone humble branch)
Package version: latest

 I just cloned the humble branch, launched the bringup launch file along with slam and nav2 as true and was breaking over this issue for 2 days.
[image]

[image]

@Srimathi-tw's fix helped me. This is after adding   ` v_samples="16" v_res="1.0" v_min_angle="-0.1" v_max_angle="0.1" `

[image]
[image]

AS-HARI, 2025-04-21:
@hilary-luo