turtlebot/turtlebot4_simulator #21: Bot not moving inside rviz for different world apart from dpot

Srimathi-tw, 2022-09-22:
**Please provide the following information:**
 - OS: Ubuntu 20.04
 - ROS Distro: Galactic
 - Built from source or installed: Source
 - Package version: Latest (irobot_create_msgs - galactic branch rest all from dependencies)

 **Expected behaviour**
On running
```ros2 launch turtlebot4_ignition_bringup ignition.launch.py model:=lite rviz:=true slam:=sync```
and moving the bot around we should able to move it around in gazebo and the same should be reflected in rviz as well

 **Actual behaviour**
We are able to move it around in gazebo but in rviz it is struck at the same dock for worlds apart from depot

**To Reproduce**
Provide the steps to reproduce:
1. Run ```ros2 launch turtlebot4_ignition_bringup ignition.launch.py model:=lite rviz:=true slam:=sync```
2. Try driving the bot [ https://turtlebot.github.io/turtlebot4-user-manual/tutorials/driving.html]( https://turtlebot.github.io/turtlebot4-user-manual/tutorials/driving.html ) by following the manual

**Other notes**
We raised an issue #20 for rplidar but after increasing the lidar value it was showing an improved range. But the same was not visible in rviz for worlds apart from depot

Srimathi-tw, 2022-09-22:
We are also experiencing the same behavior as mentioned above and thanks for the workaround we are now able to move it around in rviz

harishkumarb-tw, 2023-01-25:
@roni-kreinin The problem is that the laser scan and other sensor data have incorrect topic remapping between ros2 and ign topics. The current Implementation takes the world name from the SDF file inside <world name=> tag and uses it in the topic name.

I resolved it by editing the world/sdf file to the name which is used to call the world launch parameter **world:=** (Name of the custom world/sdf file).

Now, the topic remapping is proper and the ign data is getting published on ros2 topics.

closed 2023-09-13 by hilary-luo