turtlebot/turtlebot4_simulator #22: Turtlebot 4 without wheels and gazebo doesnt show anything

MFHuniovi, 2022-10-11:
Hello, I tried to run the turtlebot 4 simulation following the manual and could not do anything. According to rviz my robot didnt has wheels but i didnt modify anything. I was reading that it is possible that the "state publisher" doesnt have it configured correctly but I do not know where to modify it and for lastly I executed turtlebot4 diagnostic and the results was the wheels are ok.

Thanks 

**Please provide the following information:**
 - OS: **Ubuntu 20.04**
 - ROS Distro: **Galactic**
 - Built from **installed**
 - Package version: 
 
ros2@ros2-VirtualBox:~$ sudo dpkg -s ros-galactic-turtlebot4-simulator 
Package: ros-galactic-turtlebot4-simulator
Status: install ok installed
Priority: optional
Section: misc
Installed-Size: 41
Maintainer: rkreinin <rkreinin@clearpathrobotics.com>
Architecture: amd64
**Version: 0.1.1-1focal.20220803.133737**
Depends: ros-galactic-turtlebot4-ignition-bringup, ros-galactic-turtlebot4-ignition-gui-plugins, ros-galactic-turtlebot4-ignition-toolbox, ros-galactic-ros-workspace
Description: TODO: Package description

ros2@ros2-VirtualBox:~$ sudo dpkg -s ros-galactic-irobot-create-nodes
Package: ros-galactic-irobot-create-nodes
Status: install ok installed
Priority: optional
Section: misc
Installed-Size: 7488
Maintainer: Ekumen <ekumen@irbt.onmicrosoft.com>
Architecture: amd64
**Version: 1.0.3-1focal.20220730.030957**
Depends: libc6 (>= 2.14), libgcc-s1 (>= 3.0), libstdc++6 (>= 9), ros-galactic-console-bridge-vendor, libboost-all-dev, ros-galactic-angles, ros-galactic-control-msgs, ros-galactic-geometry-msgs, ros-galactic-irobot-create-msgs,
ros-galactic-irobot-create-toolbox, ros-galactic-nav-msgs, ros-galactic-rclcpp, ros-galactic-rclcpp-action, ros-galactic-rclcpp-components, ros-galactic-sensor-msgs, ros-galactic-tf2, 
ros-galactic-tf2-geometry-msgs, ros-galactic-tf2-ros, ros-galactic-ros-workspace
Description: ROS 2 Nodes for the simulated iRobot(R) Create(R) 3 Educational Robot.

 **Actual behaviour**
 Gazebo view:
[image]
[image]

rviz view:
[image]

diagnostic view:
[image]

**To Reproduce**
Provide the steps to reproduce:
1. Instalation ROS2 from `https://docs.ros.org/en/galactic/Installation/Ubuntu-Install-Debians.html`
1.1 Step Set locale
1.2 StepSetup source
1.3 Step Install ros2 desktop
1.4 Step Environment setup
1.5 Step Try some examples
2. Instalation Turtlebot 4 simulator from `https://turtlebot.github.io/turtlebot4-user-manual/software/turtlebot4_packages.html#installation-3`
2.1 Step Dev Tools
2.2 Step Ignition Edifice
2.3 Step Debian Instalation
3. Execute ros2 launch turtlebot4_ignition_bringup ignition.launch.py slam:=sync nav2:=true rviz:=true

**Other notes**
Is it normal the ignition gazebo is so slow?? My VM has 8gb RAM.

roni-kreinin, 2022-10-14:
@MFHuniovi Can you try running it again without slam and nav2? Like this `ros2 launch turtlebot4_ignition_bringup ignition.launch.py rviz:=true`. 

Also the simulation tends to be GPU intensive, so if you are running this in a VM with limited GPU resources then the performance will suffer.

MFHuniovi, 2022-10-18:
I have the same result, I will try to install drivers to use the GPU but I have a question. Why does rviz says "Warning: Invalid frame ID "right_wheel" passed to canTransform argument source_frame - frame does not exist" if the diagnostic doesn't say the same?

MFHuniovi, 2022-10-21:
Hello, I gave all my CPU cores because I read that it is very hard to use GPU in a VB, so now it works fine, but I still have the same problem with wheels. Turtlebot4 doesn't move.

[image]

roni-kreinin, 2022-11-03:
Sorry for the delay. 

> I have the same result, I will try to install drivers to use the GPU but I have a question. Why does rviz says "Warning: Invalid frame ID "right_wheel" passed to canTransform argument source_frame - frame does not exist" if the diagnostic doesn't say the same?

This is a bug with the Create 3 sim that should not actually affect anything.

> Hello, I gave all my CPU cores because I read that it is very hard to use GPU in a VB, so now it works fine, but I still have the same problem with wheels. Turtlebot4 doesn't move.

Can you provide the terminal output when you run ignition bringup?

ipa-rar, 2022-11-08:
@roni-kreinin @MFHuniovi I faced the same issue. 

I reported the issue [here](https://github.com/iRobotEducation/create3_sim/issues/190) as I felt the issue is from the **create_sim** package. 
[image]

ipa-rar, 2022-11-08:
@roni-kreinin 
> Can you provide the terminal output when you run ignition bringup?

````
[INFO] [launch]: All log files can be found below /home/ipa/.ros/log/2022-11-08-20-42-32-739058-cloud-server-29271
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ign gazebo-1]: process started with pid [29280]
[INFO] [parameter_bridge-2]: process started with pid [29282]
[INFO] [parameter_bridge-3]: process started with pid [29285]
[INFO] [parameter_bridge-4]: process started with pid [29287]
[INFO] [parameter_bridge-5]: process started with pid [29289]
[INFO] [parameter_bridge-6]: process started with pid [29291]
[INFO] [parameter_bridge-7]: process started with pid [29293]
[INFO] [parameter_bridge-8]: process started with pid [29295]
[INFO] [parameter_bridge-9]: process started with pid [29298]
[INFO] [parameter_bridge-10]: process started with pid [29314]
[INFO] [parameter_bridge-11]: process started with pid [29350]
[INFO] [parameter_bridge-12]: process started with pid [29386]
[INFO] [parameter_bridge-13]: process started with pid [29395]
[INFO] [parameter_bridge-14]: process started with pid [29407]
[INFO] [rviz2-15]: process started with pid [29409]
[INFO] [robot_state_publisher-16]: process started with pid [29416]
[INFO] [robot_state_publisher-17]: process started with pid [29420]
[INFO] [static_transform_publisher-18]: process started with pid [29454]
[INFO] [create-19]: process started with pid [29459]
[INFO] [create-20]: process started with pid [29478]
[INFO] [spawner-21]: process started with pid [29505]
[INFO] [hazards_vector_publisher-22]: process started with pid [29545]
[INFO] [ir_intensity_vector_publisher-23]: process started with pid [29562]
[INFO] [motion_control-24]: process started with pid [29572]
[INFO] [wheel_status_publisher-25]: process started with pid [29592]
[INFO] [mock_publisher-26]: process started with pid [29601]
[INFO] [robot_state-27]: process started with pid [29620]
[INFO] [kidnap_estimator_publisher-28]: process started with pid [29624]
[INFO] [ui_mgr-29]: process started with pid [29641]
[INFO] [pose_republisher_node-30]: process started with pid [29653]
[INFO] [sensors_node-31]: process started with pid [29664]
[INFO] [interface_buttons_node-32]: process started with pid [29673]
[INFO] [turtlebot4_node-33]: process started with pid [29685]
[INFO] [turtlebot4_ignition_hmi_node-34]: process started with pid [29688]
[INFO] [component_container-35]: process started with pid [29694]
[INFO] [static_transform_publisher-36]: process started with pid [29696]
[INFO] [static_transform_publisher-37]: process started with pid [29742]
[robot_state_publisher-16] The root link base_link has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
[robot_state_publisher-16] Link bump_front_center had 0 children
[robot_state_publisher-16] Link bump_front_left had 0 children
[robot_state_publisher-16] Link bump_front_right had 0 children
[robot_state_publisher-16] Link bump_left had 0 children
[robot_state_publisher-16] Link bump_right had 0 children
[robot_state_publisher-16] Link button_1 had 0 children
[robot_state_publisher-16] Link button_2 had 0 children
[robot_state_publisher-16] Link button_power had 0 children
[robot_state_publisher-16] Link cliff_front_left had 0 children
[robot_state_publisher-16] Link cliff_front_right had 0 children
[robot_state_publisher-16] Link cliff_side_left had 0 children
[robot_state_publisher-16] Link cliff_side_right had 0 children
[robot_state_publisher-16] Link front_caster_link had 0 children
[robot_state_publisher-16] Link front_left_bottom_weight_block had 0 children
[robot_state_publisher-16] Link front_left_top_weight_block had 0 children
[robot_state_publisher-16] Link front_right_bottom_weight_block had 0 children
[robot_state_publisher-16] Link front_right_top_weight_block had 0 children
[robot_state_publisher-16] Link imu_link had 0 children
[robot_state_publisher-16] Link ir_intensity_front_center_left had 0 children
[robot_state_publisher-16] Link ir_intensity_front_center_right had 0 children
[robot_state_publisher-16] Link ir_intensity_front_left had 0 children
[robot_state_publisher-16] Link ir_intensity_front_right had 0 children
[robot_state_publisher-16] Link ir_intensity_left had 0 children
[robot_state_publisher-16] Link ir_intensity_right had 0 children
[robot_state_publisher-16] Link ir_intensity_side_left had 0 children
[robot_state_publisher-16] Link ir_omni had 0 children
[... 310 lines cut ...]
[ign gazebo-1] [1;32m[Msg] [0m[1;32mFound no publishers on /clock, adding root clock topic[0m[1;32m[0m
[ign gazebo-1] [1;36m[Dbg] [SimulationRunner.cc:521] [0m[1;36mCreating PostUpdate worker threads: [0m[1;36m2[0m[1;36m[0m
[ign gazebo-1] [1;36m[Dbg] [SimulationRunner.cc:534] [0m[1;36mCreating postupdate worker thread ([0m[1;36m0[0m[1;36m)[0m[1;36m[0m
[ign gazebo-1] [1;36m[Dbg] [EntityComponentManager.cc:767] [0m[1;36mUsing components of type [[0m[1;36m273039548655896271[0m[1;36m] / [[0m[1;36mign_gazebo_components.Sensor[0m[1;36m].[0m
[ign gazebo-1] [1;36m[Dbg] [EntityComponentManager.cc:767] [0m[1;36mUsing components of type [[0m[1;36m1301926277928136113[0m[1;36m] / [[0m[1;36mign_gazebo_components.GpuLidar[0m[1;36m].[0m
[... cut ...]
[ign gazebo-1] [INFO] [1667936557.017974132] [controller_manager]: Loading controller 'joint_state_broadcaster'
[spawner-21] [INFO] [1667936557.051782272] [spawner_joint_state_broadcaster]: [94mLoaded joint_state_broadcaster[0m
[ign gazebo-1] [INFO] [1667936557.055367935] [controller_manager]: Configuring controller 'joint_state_broadcaster'
[ign gazebo-1] [INFO] [1667936557.055571593] [joint_state_broadcaster]: 'joints' or 'interfaces' parameter is empty. All available state interfaces will be published
````

ipa-rar, 2022-11-08:
TF2 frames does not contains left and right wheels
[frames.pdf](https://github.com/turtlebot/turtlebot4_simulator/files/9964756/frames.pdf)

closed 2023-09-13 by hilary-luo