turtlebot/turtlebot4_simulator #23: Gazebo Sim Rendering Not Updating Even Though Sim Does Seem to Be Running

tshumay, 2022-10-24:
**Please provide the following information:**
 - OS:Ubuntu 20.04
 - ROS Distro: Galactic
 - Built from source or installed: installed
 - Package version: 
 
Package: ros-galactic-turtlebot4-simulator
Status: install ok installed
Priority: optional
Section: misc
Installed-Size: 41
Maintainer: rkreinin <rkreinin@clearpathrobotics.com>
Architecture: amd64
Version: 0.1.1-1focal.20221016.143508
Depends: ros-galactic-turtlebot4-ignition-bringup, ros-galactic-turtlebot4-ignition-gui-plugins, ros-galactic-turtlebot4-ignition-toolbox, ros-galactic-ros-workspace
Description: TODO: Package description

 **Expected behaviour**
When I launch the Gazebo Simulator, and start the simulation, I expect that posting cmd_vel events, or running NAV via RVIZ should cause the gazebo sim to update as well. Also, the robot looks like it's maybe being rendered as a collision volume instead of the actual geometry? (Screenshot below) 

 **Actual behaviour**
 The actual robot simulation appears to be running. If I echo the /wheel_vels topic, for example, I see the wheel velocities changing as I would expect them to. Also, if I am running RVIZ with NAV goals, the RVIZ side of the simulation will run and I can see the robot moving around, but Gazebo stays stuck in place. 

**To Reproduce**
Provide the steps to reproduce:

The exact parameters to reproduce are not critical - I tried with and without RVIZ, etc. but can't get the robot to move. In the case where the sim is running via RVIZ Nav, I ran:

ros2 launch turtlebot4_ignition_bringup ignition.launch.py slam:=sync nav2:=true rviz:=true

**Other notes**
Unfortunately, new to all this, so I don't know other details that may be relevant, but I can look things up if clarification is needed. 

[image]

tshumay, 2022-11-01:
I'm running it on a Linux VM in Parallels on a Mac. It *should* have GPU support - but it is in a VM.

tshumay, 2022-11-03:
I should note, it doesn't appear to be a performance issue at least as far as the renderer is considered. The framerate is good, and I can freely move about the environment, but nothing seems to be happening with the robot on the screen.

roni-kreinin, 2022-11-03:
Can you run just `ros2 launch turtlebot4_ignition_bringup ignition.launch.py` and provide the output?

tshumay, 2022-11-05:
For sure - thanks for having a look. Here's what I've got on the output side:

trentshumay@ubuntu:~/Dev$ ros2 launch turtlebot4_ignition_bringup ignition.launch.py
[INFO] [launch]: All log files can be found below /home/trentshumay/.ros/log/2022-11-05-06-59-22-249600-ubuntu-23897
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ign gazebo-1]: process started with pid [23907]
[INFO] [parameter_bridge-2]: process started with pid [23910]
[INFO] [parameter_bridge-3]: process started with pid [23912]
[INFO] [parameter_bridge-4]: process started with pid [23914]
[INFO] [parameter_bridge-5]: process started with pid [23916]
[INFO] [parameter_bridge-6]: process started with pid [23918]
[INFO] [parameter_bridge-7]: process started with pid [23920]
[INFO] [parameter_bridge-8]: process started with pid [23922]
[INFO] [parameter_bridge-9]: process started with pid [23945]
[INFO] [parameter_bridge-10]: process started with pid [24001]
[INFO] [parameter_bridge-11]: process started with pid [24007]
[INFO] [parameter_bridge-12]: process started with pid [24025]
[INFO] [parameter_bridge-13]: process started with pid [24027]
[INFO] [parameter_bridge-14]: process started with pid [24045]
[INFO] [robot_state_publisher-15]: process started with pid [24081]
[INFO] [robot_state_publisher-16]: process started with pid [24093]
[INFO] [static_transform_publisher-17]: process started with pid [24109]
[INFO] [create-18]: process started with pid [24124]
[INFO] [create-19]: process started with pid [24154]
[INFO] [spawner-20]: process started with pid [24157]
[INFO] [hazards_vector_publisher-21]: process started with pid [24159]
[INFO] [ir_intensity_vector_publisher-22]: process started with pid [24180]
[INFO] [motion_control-23]: process started with pid [24186]
[INFO] [wheel_status_publisher-24]: process started with pid [24240]
[INFO] [mock_publisher-25]: process started with pid [24243]
[INFO] [robot_state-26]: process started with pid [24254]
[INFO] [kidnap_estimator_publisher-27]: process started with pid [24270]
[INFO] [ui_mgr-28]: process started with pid [24277]
[INFO] [pose_republisher_node-29]: process started with pid [24280]
[INFO] [sensors_node-30]: process started with pid [24290]
[INFO] [interface_buttons_node-31]: process started with pid [24339]
[INFO] [turtlebot4_node-32]: process started with pid [24343]
[INFO] [turtlebot4_ignition_hmi_node-33]: process started with pid [24362]
[INFO] [component_container-34]: process started with pid [24366]
[INFO] [static_transform_publisher-35]: process started with pid [24377]
[INFO] [static_transform_publisher-36]: process started with pid [24398]
[robot_state_publisher-15] The root link base_link has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
[robot_state_publisher-15] Link bump_front_center had 0 children
[robot_state_publisher-15] Link bump_front_left had 0 children
[robot_state_publisher-15] Link bump_front_right had 0 children
[robot_state_publisher-15] Link bump_left had 0 children
[robot_state_publisher-15] Link bump_right had 0 children
[robot_state_publisher-15] Link button_1 had 0 children
[robot_state_publisher-15] Link button_2 had 0 children
[robot_state_publisher-15] Link button_power had 0 children
[robot_state_publisher-15] Link cliff_front_left had 0 children
[robot_state_publisher-15] Link cliff_front_right had 0 children
[robot_state_publisher-15] Link cliff_side_left had 0 children
[robot_state_publisher-15] Link cliff_side_right had 0 children
[robot_state_publisher-15] Link front_caster_link had 0 children
[robot_state_publisher-15] Link front_left_bottom_weight_block had 0 children
[robot_state_publisher-15] Link front_left_top_weight_block had 0 children
[robot_state_publisher-15] Link front_right_bottom_weight_block had 0 children
[robot_state_publisher-15] Link front_right_top_weight_block had 0 children
[robot_state_publisher-15] Link imu_link had 0 children
[robot_state_publisher-15] Link ir_intensity_front_center_left had 0 children
[robot_state_publisher-15] Link ir_intensity_front_center_right had 0 children
[robot_state_publisher-15] Link ir_intensity_front_left had 0 children
[robot_state_publisher-15] Link ir_intensity_front_right had 0 children
[robot_state_publisher-15] Link ir_intensity_left had 0 children
[robot_state_publisher-15] Link ir_intensity_right had 0 children
[robot_state_publisher-15] Link ir_intensity_side_left had 0 children
[robot_state_publisher-15] Link ir_omni had 0 children
[robot_state_publisher-15] Link mouse had 0 children
[robot_state_publisher-15] Link shell_link had 7 children
[... 460 lines cut ...]
[ign gazebo-1] [Dbg] [GpuLidarSensor.cc:149] Lidar points for [turtlebot4::base_link::cliff_front_right] advertised on [world/depot/model/turtlebot4/link/base_link/sensor/cliff_front_right/scan/points]
[ign gazebo-1] [Dbg] [Lidar.cc:122] Laser scans for [turtlebot4::base_link::cliff_side_left] advertised on [world/depot/model/turtlebot4/link/base_link/sensor/cliff_side_left/scan]
[ign gazebo-1] [Dbg] [GpuLidarSensor.cc:149] Lidar points for [turtlebot4::base_link::cliff_side_left] advertised on [world/depot/model/turtlebot4/link/base_link/sensor/cliff_side_left/scan/points]
[ign gazebo-1] [Dbg] [Lidar.cc:122] Laser scans for [turtlebot4::base_link::cliff_side_right] advertised on [world/depot/model/turtlebot4/link/base_link/sensor/cliff_side_right/scan]
[... cut ...]
[ign gazebo-1] [Dbg] [Lidar.cc:122] Laser scans for [standard_dock::green_buoy_link::green_buoy] advertised on [world/depot/model/standard_dock/link/green_buoy_link/sensor/green_buoy/scan]
[ign gazebo-1] [Wrn] [OgreGpuRays.cc:253] Horizontal FOV for GPU rays is capped at 180 degrees.
[ign gazebo-1] [GUI] [Wrn] [ColladaLoader.cc:2394] Triangle input semantic: 'COLOR' is currently not supported
[ign gazebo-1] [GUI] [Wrn] [ColladaLoader.cc:2394] Triangle input semantic: 'COLOR' is currently not supported

hilary-luo, 2023-09-13:
This issue is being closed due to inactivity. If you are still experiencing the issue, feel free to reopen this ticket when you are ready to continue the troubleshooting process.

closed 2023-09-13 by hilary-luo

dlitvak, 2023-09-14:
Experiencing the same issue.  Robot doesn't move on arrow press, even though the commands do publish in /cmd_vel.
Have you been able to resolve this?  How do I debug?

I start robot via ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py model:=lite
I can see the robot.

ros2 launch turtlebot4_ignition_bringup ignition.launch.py
The scene is rendered, but the robot is not there.

hilary-luo, 2023-09-14:
@dlitvak Please open your own issue and I can help you troubleshoot there. That being said, first make sure that you are pressing play on the simulation. The operation you are describing is also what I would expect to see if you haven't pressed play. 

> I start robot via ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py model:=lite
I can see the robot.
>
> ros2 launch turtlebot4_ignition_bringup ignition.launch.py
The scene is rendered, but the robot is not there.

What you described here is expected operation in humble. The first launch file is the one that you should be using, the second one launches only the ignition gazebo environment. I see a place where this command on the website wasn't updated for Humble, I will make sure that gets fixed. The [Turtlebot 4 Simulator](https://turtlebot.github.io/turtlebot4-user-manual/software/turtlebot4_simulator.html) page and the [Tutorials](https://turtlebot.github.io/turtlebot4-user-manual/tutorials/generate_map.html) have all been updated so they are good resources on how to launch the simulation in Humble.