turtlebot/turtlebot4_simulator #32: Support for humble?

TheConstructAi, 2023-01-03:
Is there any way to make this work in humble?

roni-kreinin, 2023-01-03:
We are working on it but there are some dependencies issues currently. See https://github.com/iRobotEducation/create3_sim/issues/195

MatteoCarlone, 2023-03-07:
Do you have any news about humble simulation release? It would be really nice to try namespacing in sim :)

roni-kreinin, 2023-03-07:
@MatteoCarlone Working on some final bug fixes before releasing. You can build it from source using the `humble` branch right now if you want.

ruffsl, 2023-03-09:
Are the packages still named using the ignition convention, or will they be renamed to gazebo given the rebranding migration?

https://gazebosim.org/docs/garden/migration_from_ignition

roni-kreinin, 2023-03-09:
@ruffsl We are keeping `ign` for now as fortress supports both. Under the hood most of the `ign` packages just run the equivalent `gz` package at this point.

euivmar, 2023-04-27:
Do you have any news about the release date of the humble simulator? Do you know if the humble branch to install from source is stable or in beta version?

roni-kreinin, 2023-04-27:
@euivmar Once https://github.com/iRobotEducation/create3_sim/pull/207 is merged, we will merge our multi-robot  branches and do a release.

In the meantime you can build it from source and it should work fine. If you want multiple robot support you can use these branches:

- https://github.com/roni-kreinin/create3_sim/tree/multi_robot
- https://github.com/turtlebot/turtlebot4_simulator/tree/roni-kreinin/multi_robot
- https://github.com/turtlebot/turtlebot4/tree/roni-kreinin/multi_robot

euivmar, 2023-04-28:
@roni-kreinin  Thank you for your quick and complete answer. I hope your pull request can be reviewed and approved soon. 

Thanks again for your work.

Fran-FC, 2023-05-17:
Hello I just updated and compiled the latest changes in the humble branch. But I can't make it work. 

After launching with `ros2 launch turtlebot4_ignition_bringup ignition.launch.py` I get the following output:

```
[INFO] [launch]: All log files can be found below /home/fran/.ros/log/2023-05-17-12-40-25-355664-fran-msi-20763
[INFO] [launch]: Default logging verbosity is set to INFO
ign_args is deprecated, migrate to gz_args!
[INFO] [ruby $(which ign) gazebo-1]: process started with pid [20764]
[INFO] [parameter_bridge-2]: process started with pid [20766]
[parameter_bridge-2] [INFO] [1684320025.540771345] [clock_bridge]: Creating GZ->ROS Bridge: [/clock (ignition.msgs.Clock) -> /clock (rosgraph_msgs/msg/Clock)] (Lazy 0)
[ruby $(which ign) gazebo-1] [Wrn] [FuelClient.cc:1964] The fuel.ignitionrobotics.org URL is deprecrated. Pleasse change https://fuel.ignitionrobotics.org/1.0/OpenRobotics/models/Warehouse to https://fuel.gazebosim.org/1.0/OpenRobotics/models/Warehouse
[ruby $(which ign) gazebo-1] [Wrn] [FuelClient.cc:1964] The fuel.ignitionrobotics.org URL is deprecrated. Pleasse change https://fuel.ignitionrobotics.org/1.0/MovAi/models/pallet_box_mobile to https://fuel.gazebosim.org/1.0/MovAi/models/pallet_box_mobile
[ruby $(which ign) gazebo-1] [Wrn] [FuelClient.cc:1964] The fuel.ignitionrobotics.org URL is deprecrated. Pleasse change https://fuel.ignitionrobotics.org/1.0/MovAi/models/shelf_big to https://fuel.gazebosim.org/1.0/MovAi/models/shelf_big
[ruby $(which ign) gazebo-1] [Wrn] [FuelClient.cc:1964] The fuel.ignitionrobotics.org URL is deprecrated. Pleasse change https://fuel.ignitionrobotics.org/1.0/MovAi/models/shelf_big to https://fuel.gazebosim.org/1.0/MovAi/models/shelf_big
[ruby $(which ign) gazebo-1] [Wrn] [FuelClient.cc:1964] The fuel.ignitionrobotics.org URL is deprecrated. Pleasse change https://fuel.ignitionrobotics.org/1.0/MovAi/models/shelf_big to https://fuel.gazebosim.org/1.0/MovAi/models/shelf_big
[ruby $(which ign) gazebo-1] [Wrn] [FuelClient.cc:1964] The fuel.ignitionrobotics.org URL is deprecrated. Pleasse change https://fuel.ignitionrobotics.org/1.0/MovAi/models/shelf to https://fuel.gazebosim.org/1.0/MovAi/models/shelf
[ruby $(which ign) gazebo-1] [Wrn] [FuelClient.cc:1964] The fuel.ignitionrobotics.org URL is deprecrated. Pleasse change https://fuel.ignitionrobotics.org/1.0/MovAi/models/shelf to https://fuel.gazebosim.org/1.0/MovAi/models/shelf
[ruby $(which ign) gazebo-1] [Wrn] [FuelClient.cc:1964] The fuel.ignitionrobotics.org URL is deprecrated. Pleasse change https://fuel.ignitionrobotics.org/1.0/MovAi/models/shelf to https://fuel.gazebosim.org/1.0/MovAi/models/shelf
[ruby $(which ign) gazebo-1] [GUI] [Wrn] [Application.cc:797] [QT] qrc:/qml/StyleDialog.qml:112:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[ruby $(which ign) gazebo-1] [GUI] [Wrn] [Application.cc:797] [QT] qrc:/qml/StyleDialog.qml:105:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[ruby $(which ign) gazebo-1] [GUI] [Wrn] [Application.cc:797] [QT] qrc:/qml/StyleDialog.qml:98:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[ruby $(which ign) gazebo-1] [GUI] [Wrn] [Application.cc:797] [QT] qrc:qml/Main.qml:102:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[ruby $(which ign) gazebo-1] [GUI] [Wrn] [Application.cc:797] [QT] qrc:/qml/PluginMenu.qml:27:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[ruby $(which ign) gazebo-1] [Wrn] [FuelClient.cc:1964] The fuel.ignitionrobotics.org URL is deprecrated. Pleasse change https://fuel.ignitionrobotics.org/1.0/MovAi/models/shelf to https://fuel.gazebosim.org/1.0/MovAi/models/shelf
[ruby $(which ign) gazebo-1] [GUI] [Wrn] [Application.cc:797] [QT] file::/Gazebo/GazeboDrawer.qml:241:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[ruby $(which ign) gazebo-1] [Wrn] [FuelClient.cc:1964] The fuel.ignitionrobotics.org URL is deprecrated. Pleasse change https://fuel.ignitionrobotics.org/1.0/MovAi/models/shelf to https://fuel.gazebosim.org/1.0/MovAi/models/shelf
[ruby $(which ign) gazebo-1] [GUI] [Wrn] [Application.cc:797] [QT] file::/Gazebo/GazeboDrawer.qml:147:3: QML Dialog: Binding loop detected for property "implicitHeight"
[ruby $(which ign) gazebo-1] [GUI] [Wrn] [Application.cc:797] [QT] file::/Gazebo/GazeboDrawer.qml:147:3: QML Dialog: Binding loop detected for property "implicitHeight"
[ruby $(which ign) gazebo-1] [Wrn] [FuelClient.cc:1964] The fuel.ignitionrobotics.org URL is deprecrated. Pleasse change https://fuel.ignitionrobotics.org/1.0/MovAi/models/shelf_big to https://fuel.gazebosim.org/1.0/MovAi/models/shelf_big
[ruby $(which ign) gazebo-1] [Wrn] [FuelClient.cc:1964] The fuel.ignitionrobotics.org URL is deprecrated. Pleasse change https://fuel.ignitionrobotics.org/1.0/MovAi/models/shelf_big to https://fuel.gazebosim.org/1.0/MovAi/models/shelf_big
[... cut ...] le_Diffuse.jpg to https://fuel.gazebosim.org/1.0/openrobotics/models/table/3/files/Table_Diffuse.jpg
[ruby $(which ign) gazebo-1] [GUI] [Wrn] [FuelClient.cc:1964] The fuel.ignitionrobotics.org URL is deprecrated. Pleasse change https://fuel.ignitionrobotics.org/1.0/openrobotics/models/table/3/files/Table_Diffuse.jpg to https://fuel.gazebosim.org/1.0/openrobotics/models/table/3/files/Table_Diffuse.jpg
```

I also can't view any topics published and the robot does not appear in the world.

[image]

roni-kreinin, 2023-05-17:
@Fran-FC The debian for the simulator is now released on the ros2 testing server. Try installing it from there.

To use the testing server, edit `/etc/apt/sources.list.d/ros2.list` and use `http://packages.ros.org/ros2-testing/ubuntu` as the package server. Then call `sudo apt update && sudo apt install ros-humble-turtlebot4-simulator` to install it.

tommy4111, 2023-05-18:
Anyone figured out how to spawn turtlebot?   I'm using testing server.

Fran-FC, 2023-05-18:
> @Fran-FC The debian for the simulator is now released on the ros2 testing server. Try installing it from there.
>
> To use the testing server, edit `/etc/apt/sources.list.d/ros2.list` and use [...]

Thanks I installed the package via apt but the exact same errors happen, the robot does not spawn and I get the same output in the terminal.

roni-kreinin, 2023-05-18:
@Fran-FC I just noticed you are using the `ignition.launch.py` launch file. That one just launches the sim and spawns the world. Try using `ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py`

@tommy4111 `ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py`

roni-kreinin, 2023-05-18:
Documentation will be up soon for the sim.

Fran-FC, 2023-05-23:
Thanks, everything is correct, but the simulation performance is very poor. I don't know if it is due to my PC specs, these are the following:
- i7 4 cores (8 threads) 1.8GHz
- Nvidia GTX 1050
- 16Gb RAM

The simulation goes to 8% of the real time.

zenas91, 2023-05-25:
@roni-kreinin I am having similar issue. Whenever I run `turtlebot4_ignition.launch.py`, the world starts but the robot is not spawn. Additionally, I get this error `[ERROR] [launch]: Caught exception in launch (see debug for traceback): No module named 'irobot_create_common_bringup'`. Running `ignition.launch.py` gives me similar respond as [this](https://github.com/turtlebot/turtlebot4_simulator/issues/32#issuecomment-1551163970)

zenas91, 2023-05-25:
> > @roni-kreinin I am having similar issue. Whenever I run `turtlebot4_ignition.launch.py`, the world starts but the robot is not spawn. Additionally, I get this error `[ERROR] [launch]: Caught exception in launch (see debug for traceback): No module named 'irobot_create_common_bringup'`. Running `ignition.launch.py` gives similar respond as [this](https://github.com/turtlebot/turtlebot4_simulator/issues/32#issuecomment-1551163970)
> 
> I solved the problem of the module not found by compiling the create3_sim repo, it seems that the installation of the turtlebot4_sim package does not install this dependency.

@Fran-FC Did you do that in the same workspace or you created a separate workspace for create3?

Fran-FC, 2023-05-25:
> @Fran-FC Did you do that in the same workspace or you created a separate workspace for create3?

I compiled it all on the same ws

zenas91, 2023-05-25:
> I compiled it all on the same ws

Thanks

Fran-FC, 2023-05-25:
The base simulation seems to work fine, but the problem comes when launching SLAM. It seems to be the same problem as referenced in this issue https://github.com/turtlebot/turtlebot4/issues/134#issuecomment-1531270408 

The output of the SLAM is the following:

```
[sync_slam_toolbox_node-1] [INFO] [1685024243.971329229] [slam_toolbox]: Message Filter dropping message: frame 'turtlebot4/rplidar_link/rplidar' at time 16.770 for reason 'discarding message because the queue is full'
```
The TF tree seems fine though:

[frames_2023-05-25_16.23.30.pdf](https://github.com/turtlebot/turtlebot4_simulator/files/11565952/frames_2023-05-25_16.23.30.pdf)

Any suggestions?

hilary-luo, 2023-09-13:
Closing due to inactivity and because Humble is released.

closed 2023-09-13 by hilary-luo