turtlebot/turtlebot4_simulator #33: Rosdep Cannot Find turtlebot4_simulator Dependencies

BashByr, 2023-02-21:
**Please provide the following information:**
 - OS: Ubuntu 20.04.5 LTS
 - ROS Distro: (e.g. Galactic): Galactic
 - Built from source or installed: Source
 - Package version: 500cfb8

 **Expected behaviour**
 Successful rosdep installation of TBot4 simulator, with all rosdep definitions found.

 **Actual behaviour**
`ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
irobot_create_control: Cannot locate rosdep definition for [ros2_controllers]
turtlebot4_ignition_toolbox: Cannot locate rosdep definition for [ros_ign_interfaces]
turtlebot4_ignition_bringup: Cannot locate rosdep definition for [ros_ign_bridge]
irobot_create_gazebo_bringup: Cannot locate rosdep definition for [gazebo_ros]
irobot_create_ignition_toolbox: Cannot locate rosdep definition for [ros_ign_interfaces]
irobot_create_common_bringup: Cannot locate rosdep definition for [xacro]
irobot_create_ignition_bringup: Cannot locate rosdep definition for [ros_ign_interfaces]
turtlebot4_viz: Cannot locate rosdep definition for [rqt_robot_monitor]
irobot_create_gazebo_plugins: Cannot locate rosdep definition for [gazebo_ros]`

**To Reproduce**
Provide the steps to reproduce:
1. cd ~/turtlebot4_ws/src
2. git clone https://github.com/turtlebot/turtlebot4_simulator.git
3. cd ~/turtlebot4_ws
4. vcs import src < src/turtlebot4_simulator/dependencies.repos
5. source /opt/ros/galactic/setup.bash
6. rosdep install --from-path src -yi --rosdistro=galactic

**Other notes**
I thought this might’ve been related to #24, but I noticed that the latest commit changed the `dependencies.repos` file to include the galactic branches specifically. To be sure of that, I manually cloned the galactic branch of every repo, but the same error popped up.

I also noticed that the main branch for those repos (example: https://github.com/gazebosim/ros_gz/tree/ros2), have moved beyond galactic, and the package names have changed (example: ros_ign_interfaces is now ros_gz_interfaces). I’m not sure how to have rosdep look at definitions for galactic specifically.

roni-kreinin, 2023-02-22:
@BashByr The `dependencies.repos` file might be out of date at this point. I would recommend installing from binary by calling `sudo apt install ros-galactic-turtlebot4-simulator` which should install all dependencies. Then build whichever packages you want to modify from source.

BashByr, 2023-03-02:
@roni-kreinin understood, thanks. Side question: is there a rough date for when namespacing will become available for the TBot4 simulation? Or even the physical bots?

roni-kreinin, 2023-03-03:
@BashByr Namespacing is available on the physical robot in Humble now. The Humble images should be uploaded today, along with updates to the user manual. Multiple robot support in simulation is dependent on that feature being added to create3_sim. Don't have a timeline for that yet.

BashByr, 2023-03-05:
Gotcha, thanks a ton for the update.

closed 2023-03-05 by BashByr