turtlebot/turtlebot4_simulator #38: Turtlebot4 simulator dock/undock?

languyen14507, 2023-04-17:
Hello,

How does one make Turtlebot4 dock or undock in its simulator?

Thank you.

Jayden-F, 2023-05-05:
Hello @languyen14507, you can send the dock and undock commands via the Create3 Dock and Undock Actions: 

There is a more in depth description here:
https://iroboteducation.github.io/create3_docs/api/docking/

However to save a click here are the commands:

# Undock 
~~~
ros2 action send_goal /undock irobot_create_msgs/action/Undock "{}"
~~~

# Dock

**Galactic**
~~~
ros2 action send_goal /dock irobot_create_msgs/action/DockServo "{}"
~~~
**Humble**
~~~
ros2 action send_goal /dock irobot_create_msgs/action/Dock "{}"
~~~

languyen14507, 2023-05-14:
Can the Turtlebot 4 simulator run SLAM and Navigation like the Turtlebot 3 simulator? I have never been able to undock the Turtlebot 4 simulation successfully so I couldn't tell. The undocking process always stopped and never finished because the undock goal was always canceled by the simulator before it has change to finish. Thanks.

languyen14507, 2023-05-15:
I don't see any RTL on the simulator gazebo display but I see these messages on the terminal where I activated the simulator:
spawner-20] [INFO] [1684166862.565217662] [spawner_joint_state_broadcaster]: Configured and started joint_state_broadcaster
[parameter_bridge-3] [INFO] [1684166862.692902468] [cmd_vel_bridge]: Passing message from ROS geometry_msgs/msg/Twist to Ignition ignition.msgs.Twist (showing msg only once per type)
[INFO] [spawner-20]: process has finished cleanly [pid 50118]
[INFO] [spawner-37]: process started with pid [51358]
[ign gazebo-1] [INFO] [1684166863.133635835] [controller_manager]: Loading controller 'diffdrive_controller'
[spawner-37] [INFO] [1684166863.211148035] [spawner_diffdrive_controller]: Loaded diffdrive_controller
[ign gazebo-1] [INFO] [1684166863.212986343] [controller_manager]: Configuring controller 'diffdrive_controller'
[spawner-37] [INFO] [1684166863.321551000] [spawner_diffdrive_controller]: Configured and started diffdrive_controller
[INFO] [spawner-37]: process has finished cleanly [pid 51358]
[turtlebot4_node-32] [INFO] [1684166871.830227083] [turtlebot4_node]: Undocking
[turtlebot4_node-32] [INFO] [1684166871.830622690] [turtlebot4_node]: Waiting for undock action server
[turtlebot4_node-32] [INFO] [1684166872.831421095] [turtlebot4_node]: undock action server available, sending goal
[motion_control-23] [INFO] [1684166872.831893474] [motion_control]: Received new undock goal
[turtlebot4_node-32] [INFO] [1684166872.832237148] [turtlebot4_node]: undock goal accepted by server, waiting for result
[turtlebot4_node-32] [ERROR] [1684166880.632648365] [turtlebot4_node]: undock goal was canceled
[motion_control-23] [WARN] [1684166880.637115688] [motion_control]: Reached backup limit! Stop Driving robot backward or disable from safety_override parameter!
[motion_control-23] [WARN] [1684166953.540066712] [motion_control]: Reflex Exceeded Runtime without clearing hazard

languyen14507, 2023-05-15:
When trying the run NAV2 with the Turtlebot 4 simulator with the command
ros2 launch turtlebot4_ignition_bringup ignition.launch.py nav2:=true slam:=off localization:=true

I got the following error messages:
[ign gazebo-1] [Wrn] [OgreGpuRays.cc:253] Horizontal FOV for GPU rays is capped at 180 degrees.
[controller_server-38] [INFO] [1684169154.433378486] [local_costmap.local_costmap]: Timed out waiting for transform from base_link to odom to become available, tf error: Could not find a connection between 'odom' and 'base_link' because they are not part of the same tree.Tf has two or more unconnected trees.
[controller_server-38] [INFO] [1684169154.933372248] [local_costmap.local_costmap]: Timed out waiting for transform from base_link to odom to become available, tf error: Could not find a connection between 'odom' and 'base_link' because they are not part of the same tree.Tf has two or more unconnected trees.
[ign gazebo-1] [GUI] [Wrn] [ColladaLoader.cc:2394] Triangle input semantic: 'COLOR' is currently not supported
[ign gazebo-1] [GUI] [Wrn] [ColladaLoader.cc:2394] Triangle input semantic: 'COLOR' is currently not supported
[controller_server-38] [INFO] [1684169155.433371462] [local_costmap.local_costmap]: Timed out waiting for transform from base_link to odom to become available, tf error: Could not find a connection between 'odom' and 'base_link' because they are not part of the same tree.Tf has two or more unconnected trees.
[controller_server-38] [INFO] [1684169155.933372924] [local_costmap.local_costmap]: Timed out waiting for transform from base_link to odom to become available, tf error: Could not find a connection between 'odom' and 'base_link' because they are not part of the same tree.Tf has two or more unconnected trees.
[controller_server-38] [INFO] [1684169156.433373573] [local_costmap.local_costmap]: Timed out waiting for transform from base_link to odom to become available, tf error: Could not find a connection between 'odom' and 'base_link' because they are not part of the same tree.Tf has two or more unconnected trees.
[controller_server-38] [INFO] [1684169156.933385483] [local_costmap.local_costmap]: Timed out waiting for transform from base_link to odom to become available, tf error: Could not find a connection between 'odom' and 'base_link' because they are not part of the same tree.Tf has two or more unconnected trees.

roni-kreinin, 2023-05-15:
You should see a fluctuating percentage value at the bottom right corner of the scene. If you expand that you should see something like this: 
[image]

languyen14507, 2023-05-15:
[image]

languyen14507, 2023-05-15:
The RTF value just fluctutates around 10 and 11%

roni-kreinin, 2023-05-15:
That's quite low. The simulation is basically running at 10% real-time, which can cause issues. 

Does your PC have a dedicated GPU?

languyen14507, 2023-05-15:
It has an Nvidia GEForce GTX 1650 Mobile and an Intel UHD Graphics 630 Mobile

roni-kreinin, 2023-05-15:
If you enter `nvidia-settings` into your terminal, you should be able to go to PRIME Profiles and select NVIDIA (Performance Mode) to force the sim to use your GPU. You can also try changing the PowerMizer power mode to performance.

I think the sim is using your integrated graphics right now which is causing the poor performance.

languyen14507, 2023-05-15:
I was able to switch the laptop to using the Nvidia GPU driver but the RTF value only increases to 13-14%. The undocking is still canceled due to reaching backup limit. The NAV2 run still has the  tf error: Could not find a connection between 'odom' and 'base_link' because they are not part of the same tree.Tf has two or more unconnected trees.
[image]

roni-kreinin, 2023-05-16:
I'm running  the sim on a GTX1650Ti and get about 70% RTF so I think the sim is still using integrated graphics here. You may need to dig some more in your GPU settings.

languyen14507, 2023-05-17:
I was able to correct the GPU driver settings and the simulation ran fine after that (70-80% RTF). Thanks so much.

closed 2023-05-17 by languyen14507

CraftyCranberry, 2024-02-21:
> I was able to correct the GPU driver settings and the simulation ran fine after that (70-80% RTF). Thanks so much.

hey @languyen14507 i am having the same issue. may i know how you fixed the driver settings?

[image]

languyen14507, 2024-02-21:
Instead of selecting "from nvidia-driver-530-open" as shown in the picture above, I changed the selection to  "from nvidia-driver-418-server", clicked Apply Changes, waited until Aplly Changes is finished then reboot the computer.

aldodiaz-UFG, 2024-03-21:
> Instead of selecting "from nvidia-driver-530-open" as shown in the picture above, I changed the selection to "from nvidia-driver-418-server", clicked Apply Changes, waited until Aplly Changes is finished then reboot the computer.

I have a GeForce 940M, do you have any hint how you selected the correct driver from the list in order to fix the RTF issue?

> I'm running the sim on a GTX1650Ti and get about 70% RTF so I think the sim is still using integrated graphics here. You may need to dig some more in your GPU settings.

Do you have any driver suggestion for a GeForce 940M (about 20% RTX on average)? (see 
[image]
picture attached)

qmnguyen14507, 2024-04-06:
I would suggest to cycle through the list of available drivers until you find one that works.

exDawn, 2024-11-13:
I also encountered the same problem, has solved, thank you very much.

My original RTF was only 2%, but with Performance Mode in nvidia-settings, my RTF was around 10%. However, the RTF remains unchanged after switching the driver.

I refer to the https://cloud.tencent.com/developer/article/2046885, terminal input inxi - G, found the OpenGL use sets show here.
My computer, Intel+NVIDIA combination, is now in centralized display mode. However, after switching the single-display mode, the OpenGL of inxi-G still uses intel.
Looking up the information, I found that although I have switched to NVIDIA single display mode, the NVIDIA driver may not be enabled correctly because the current Wayland session is used. NVIDIA single display mode is more stable in X11 environment, especially in OpenGL rendering.
After logging out and re-selecting Ubuntu on Xorg, I logged in, the inxi-G display was NVIDIA, and finally the RTF was about 60%.