turtlebot/turtlebot4_simulator #40: Multirobot Turtlebot4 Crash

Jayden-F, 2023-05-05:
**Please provide the following information:**
 - OS: Ubuntu 22.04
 - ROS Distro: Humble
 - Built from source or installed: Multirobot sim
 - Package version: 

 1a0a68404324c35062b4277e19ca0ecedb4443d6 src/create3_sim (2.0.0-18-g1a0a684)
 382711f9955071ab92f52198c5461ea1ca9a79ed src/turtlebot4 (1.0.1-10-g382711f)
 f0f0a189496d78a1475d53940e597e26b5977b20 src/turtlebot4_simulator (0.1.1-23-gf0f0a18)

 **Expected behaviour**
When adding a second turtlebot4 simulation I expect the robot to spawn and be controllable.

 **Actual behaviour**

The robot will spawn and you will briefly see it in the simulation before crashing. 
~~~
[ruby $(which ign) gazebo-1] [Wrn] [OgreGpuRays.cc:258] Horizontal FOV for GPU rays is capped at 180 degrees.
[ruby $(which ign) gazebo-1] [INFO] [1683246401.231851001] [robot2.controller_manager]: Setting use_sim_time=True for joint_state_broadcaster to match controller manager (see ros2_control#325 for details)
[ruby $(which ign) gazebo-1] [Err] [SceneManager.cc:191] Visual: [ground_plane] already exists
[ruby $(which ign) gazebo-1] [Err] [SceneManager.cc:191] Visual: [warehouse] already exists
[ruby $(which ign) gazebo-1] [Err] [SceneManager.cc:191] Visual: [pallet_box_0] already exists
[ruby $(which ign) gazebo-1] [Err] [SceneManager.cc:191] Visual: [shelf_big_0] already exists
[ruby $(which ign) gazebo-1] [Err] [SceneManager.cc:191] Visual: [shelf_big_1] already exists
[ruby $(which ign) gazebo-1] [Err] [SceneManager.cc:191] Visual: [shelf_big_2] already exists
[ruby $(which ign) gazebo-1] [Err] [SceneManager.cc:191] Visual: [shelf_3] already exists
[ruby $(which ign) gazebo-1] [Err] [SceneManager.cc:191] Visual: [shelf_4] already exists
[ruby $(which ign) gazebo-1] [Err] [SceneManager.cc:191] Visual: [shelf_5] already exists
[ruby $(which ign) gazebo-1] [Err] [SceneManager.cc:191] Visual: [shelf_6] already exists
[ruby $(which ign) gazebo-1] [Err] [SceneManager.cc:191] Visual: [shelf_7] already exists
[ruby $(which ign) gazebo-1] [Err] [SceneManager.cc:191] Visual: [shelf_big_3] already exists
[ruby $(which ign) gazebo-1] [Err] [SceneManager.cc:191] Visual: [shelf_big_4] already exists
[ruby $(which ign) gazebo-1] [Err] [SceneManager.cc:191] Visual: [shelf_0] already exists
[ruby $(which ign) gazebo-1] [Err] [SceneManager.cc:191] Visual: [shelf_1] already exists
[ruby $(which ign) gazebo-1] [Err] [SceneManager.cc:191] Visual: [shelf_2] already exists
[ruby $(which ign) gazebo-1] [Err] [SceneManager.cc:191] Visual: [barrier_0] already exists
[ruby $(which ign) gazebo-1] [Err] [SceneManager.cc:191] Visual: [barrier_1] already exists
[ruby $(which ign) gazebo-1] [Err] [SceneManager.cc:191] Visual: [barrier_2] already exists
[ruby $(which ign) gazebo-1] [Err] [SceneManager.cc:191] Visual: [barrier_3] already exists
[ruby $(which ign) gazebo-1] [Err] [SceneManager.cc:191] Visual: [chair_0] already exists
[ruby $(which ign) gazebo-1] [Err] [SceneManager.cc:191] Visual: [chair_1] already exists
[ruby $(which ign) gazebo-1] [Err] [SceneManager.cc:191] Visual: [fchair_0] already exists
[ruby $(which ign) gazebo-1] [Err] [SceneManager.cc:191] Visual: [fchair1] already exists
[ruby $(which ign) gazebo-1] [Err] [SceneManager.cc:191] Visual: [table0] already exists
[ruby $(which ign) gazebo-1] [Err] [SceneManager.cc:191] Visual: [Person 0 - Sitting] already exists
[ruby $(which ign) gazebo-1] [Err] [SceneManager.cc:177] Parent entity with Id: [137] not found. Not adding model visual with ID[141]  and name [artifact_proximity_detector] to the rendering scene.
[ruby $(which ign) gazebo-1] [Err] [SceneManager.cc:191] Visual: [Person 1 - Standing] already exists
[ruby $(which ign) gazebo-1] [Err] [SceneManager.cc:191] Visual: [Person 2 - Walking] already exists
[ruby $(which ign) gazebo-1] [Err] [SceneManager.cc:191] Visual: [robot1/standard_dock] already exists
[ruby $(which ign) gazebo-1] [Err] [SceneManager.cc:191] Visual: [robot1/turtlebot4] already exists
[ruby $(which ign) gazebo-1] [Err] [SceneManager.cc:191] Visual: [robot2/standard_dock] already exists
[ruby $(which ign) gazebo-1] [Err] [SceneManager.cc:191] Visual: [robot2/turtlebot4] already exists
[ruby $(which ign) gazebo-1] terminate called after throwing an instance of 'Ogre::ItemIdentityException'
[ruby $(which ign) gazebo-1]   what():  OGRE EXCEPTION(4:ItemIdentityException): A scene node with the name sun already exists in SceneManager::createSceneNode at ./OgreMain/src/OgreSceneManager.cpp (line 840)
[ruby $(which ign) gazebo-1] Stack trace (most recent call last) in thread 449038:
[ruby $(which ign) gazebo-1] #21   Object "[0xffffffffffffffff]", at 0xffffffffffffffff, in 
[ruby $(which ign) gazebo-1] #20   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f40b45269ff, in 
[ruby $(which ign) gazebo-1] #19   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f40b4494b42, in 
[ruby $(which ign) gazebo-1] #18   Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f40afcdc2b2, in 
[ruby $(which ign) gazebo-1] #17   Object "/usr/lib/x86_64-linux-gnu/ign-gazebo-6/plugins/libignition-gazebo-sensors-system.so", at 0x7f408c1412bf, in ignition::gazebo::v6::systems::SensorsPrivate::RenderThread()
[ruby $(which ign) gazebo-1] #16   Object "/usr/lib/x86_64-linux-gnu/ign-gazebo-6/plugins/libignition-gazebo-sensors-system.so", at 0x7f408c1400e6, in ignition::gazebo::v6::systems::SensorsPrivate::RunOnce()
[ruby $(which ign) gazebo-1] #15   Object "/lib/x86_64-linux-gnu/libignition-gazebo6-rendering.so.6", at 0x7f407c76b77f, in ignition::gazebo::v6::RenderUtil::Update()
[ruby $(which ign) gazebo-1] #14   Object "/lib/x86_64-linux-gnu/libignition-gazebo6-rendering.so.6", at 0x7f407c7b1294, in ignition::gazebo::v6::SceneManager::CreateLight(unsigned long, sdf::v12::Light const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned long)
[ruby $(which ign) gazebo-1] #13   Object "/lib/x86_64-linux-gnu/libignition-rendering6.so.6", at 0x7f407c607ade, in ignition::rendering::v6::BaseScene::CreateDirectionalLight(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
[ruby $(which ign) gazebo-1] #12   Object "/lib/x86_64-linux-gnu/libignition-rendering6.so.6", at 0x7f407c60bb3c, in ignition::rendering::v6::BaseScene::CreateDirectionalLight(unsigned int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
[ruby $(which ign) gazebo-1] #11   Object "/usr/lib/x86_64-linux-gnu/ign-rendering-6/engine-plugins/libignition-rendering-ogre.so", at 0x7f407c3b4a3b, in ignition::rendering::v6::OgreScene::CreateDirectionalLightImpl(unsigned int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
[ruby $(which ign) gazebo-1] #10   Object "/usr/lib/x86_64-linux-gnu/ign-rendering-6/engine-plugins/libignition-rendering-ogre.so", at 0x7f407c3b2d97, in ignition::rendering::v6::OgreScene::InitObject(std::shared_ptr<ignition::rendering::v6::OgreObject>, unsigned int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
[ruby $(which ign) gazebo-1] #9    Object "/usr/lib/x86_64-linux-gnu/ign-rendering-6/engine-plugins/libignition-rendering-ogre.so", at 0x7f407c371441, in virtual thunk to ignition::rendering::v6::OgreLight::Init()
[ruby $(which ign) gazebo-1] #8    Object "/usr/lib/x86_64-linux-gnu/ign-rendering-6/engine-plugins/libignition-rendering-ogre.so", at 0x7f407c3a24a8, in ignition::rendering::v6::OgreNode::Init()
[ruby $(which ign) gazebo-1] #7    Object "/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0", at 0x7f4074337a52, in 
[ruby $(which ign) gazebo-1] #6    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f40afcae517, in __cxa_throw
[ruby $(which ign) gazebo-1] #5    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f40afcae2b6, in std::terminate()
[ruby $(which ign) gazebo-1] #4    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f40afcae24b, in 
[ruby $(which ign) gazebo-1] #3    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f40afca2bbd, in 
[ruby $(which ign) gazebo-1] #2    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f40b44287f2, in abort
[ruby $(which ign) gazebo-1] #1    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f40b4442475, in raise
[ruby $(which ign) gazebo-1] #0    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f40b4496a7c, in pthread_kill
[ruby $(which ign) gazebo-1] Aborted (Signal sent by tkill() 408512 1000)

~~~

**To Reproduce**

1. run `ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py model:=lite namespace:=robot1`
2. run `ros2 launch turtlebot4_ignition_bringup turtlebot4_spawn.launch.py namespace:=robot2 model:=lite y:=1.0`
3. The robot spawns then the sim imediately crashes

**Additional Info**

I am able to run the create 3 multirobot example without issues.

Jayden-F, 2023-05-05:
Adding the relevant pull request for visibility #35

Jayden-F, 2023-05-15:
I built from the newly merged multi-robot sim and it works.

closed 2023-05-17 by roni-kreinin

abdosalem490, 2023-11-17:
@Jayden-F 
how did you solve the problem?
because I am facing the same problem on same rosdistro (humble) using (gazebo fortress) on Ubunut 22.04.
I even raised an issue for this, spawning 2 robots will raise the same error messages you posted before but will not close the simulatio.
trying to spawn a third turtlebot4 will crash the simulation.
so how did you solve the problem?

msminirobot, 2023-11-28:
Same problem here; would love to know how to simulate more than two turtlebots!