turtlebot/turtlebot4_simulator #44: Requesting List of World Names in loop and gazebo gui not responding

blaiblayer, 2023-06-14:
- OS: Ubuntu 22.04.2
 - ROS Distro: Humble
 - Built from source in User PC

STEPS FOLLOWED:

I want to use turtlebot4 simulator and I have installed everything as stated in the turtlebot4 guide in my user PC. No problems arise during compilation.

LOG:

[INFO] [launch]: All log files can be found below [...]
[INFO] [launch]: Default logging verbosity is set to INFO
ign_args is deprecated, migrate to gz_args!
[INFO] [ruby $(which ign) gazebo-1]: process started with pid [107055]
[INFO] [parameter_bridge-2]: process started with pid [107057]
[INFO] [robot_state_publisher-3]: process started with pid [107060]
[INFO] [joint_state_publisher-4]: process started with pid [107063]
[INFO] [robot_state_publisher-5]: process started with pid [107065]
[INFO] [static_transform_publisher-6]: process started with pid [107067]
[INFO] [create-7]: process started with pid [107069]

[...]

[create-7] [ros_ign_gazebo] is deprecated! Redirecting to use [ros_gz_sim] instead!
[create-7] 
[create-7] Invalid partition name 
[create-7] [INFO] [1686673624.014756704] [ros_gz_sim]: Requesting list of world names.
[create-7] Service [/gazebo/worlds] is not valid.
[create-7] [INFO] [1686673624.015201605] [ros_gz_sim]: Requesting list of world names.
[create-7] Service [/gazebo/worlds] is not valid.
[create-7] [INFO] [1686673624.015302370] [ros_gz_sim]: Requesting list of world names.
[create-7] Service [/gazebo/worlds] is not valid.
[create-7] [INFO] [1686673624.015370684] [ros_gz_sim]: Requesting list of world names.
[create-7] Service [/gazebo/worlds] is not valid.
 
ERROR:

The following commands repeat on loop (and gazebo gui does shows pop up with not responding message) when launching turtlebot4_ignition.launch.py: 

[create-7] [INFO] [1686673624.015370684] [ros_gz_sim]: Requesting list of world names.
[create-7] Service [/gazebo/worlds] is not valid.

Sometimes a message appear on terminal:  GUI requesting list of world names. The server may be busy downloading resources. Please be patient.

I have waited for 30min and nothing changes, I have also disabled my firewall in case but it is not the solution.

roni-kreinin, 2023-06-14:
@blaiblayer The first time you launch gazebo it will do these background downloads. Does this persist on subsequent launches?

blaiblayer, 2023-06-15:
@roni-kreinin Yes it does persist and gazebo gui window still showing message npt responding

lizsterkenburg, 2023-09-08:
@roni-kreinin @blaiblayer I have the same problem, do either of you already know a fix for this perhaps?

teufelweich, 2023-09-08:
Same for me. 

```
[create-8] [INFO] [1694077383.011122552] [ros_gz_sim]: Requesting list of world names.
[create-7] [INFO] [1694077383.017145205] [ros_gz_sim]: Requesting list of world names.
[create-8] [INFO] [1694077388.011700932] [ros_gz_sim]: Requesting list of world names.
[create-7] [INFO] [1694077388.017577638] [ros_gz_sim]: Requesting list of world names.
[create-8] [INFO] [1694077393.012184950] [ros_gz_sim]: Requesting list of world names.
[create-7] [INFO] [1694077393.018004785] [ros_gz_sim]: Requesting list of world names.
[ruby $(which ign) gazebo-1] [Msg] Downloading model [https://fuel.ignitionrobotics.org/openrobotics/models/artifact proximity detector]
[ruby $(which ign) gazebo-1] [Dbg] [Zip.cc:180] Created file [/local/marius/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/artifact proximity detector/5/model.config]
[ruby $(which ign) gazebo-1] [Dbg] [Zip.cc:180] Created file [/local/marius/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/artifact proximity detector/5/model.sdf]
[ruby $(which ign) gazebo-1] [Msg] Downloading model [https://fuel.gazebosim.org/openrobotics/models/malevisitoronphone]
[ruby $(which ign) gazebo-1] [Dbg] [FuelClient.cc:2002] Downloading from a referral link [https://gz-fuel-resources-production.s3.amazonaws.com/OpenRobotics/d7fd6d9c-f3c4-4dab-a616-35435c07bee3/.zips/1.zip?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAS5OHIRKDEEBENL5G%2F20230907%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20230907T090314Z&X-Amz-Expires=3600&X-Amz-SignedHeaders=host&X-Amz-Signature=df20c7ffb1b5488852678cdcd7da4929c191ff0ae2f70a379bc6d16bc7bb08a2]
[ruby $(which ign) gazebo-1] [Dbg] [Zip.cc:180] Created file [/local/marius/.ignition/fuel/fuel.gazebosim.org/openrobotics/models/malevisitoronphone/1/meshes/MaleVisitor.png]
[ruby $(which ign) gazebo-1] [Dbg] [Zip.cc:180] Created file [/local/marius/.ignition/fuel/fuel.gazebosim.org/openrobotics/models/malevisitoronphone/1/meshes/MaleVisitorStatic.mtl]
[ruby $(which ign) gazebo-1] [Dbg] [Zip.cc:180] Created file [/local/marius/.ignition/fuel/fuel.gazebosim.org/openrobotics/models/malevisitoronphone/1/meshes/MaleVisitorStatic.obj]
[ruby $(which ign) gazebo-1] [Dbg] [Zip.cc:180] Created file [/local/marius/.ignition/fuel/fuel.gazebosim.org/openrobotics/models/malevisitoronphone/1/meshes/MaleVisitorStatic_Col.mtl]
[ruby $(which ign) gazebo-1] [Dbg] [Zip.cc:180] Created file [/local/marius/.ignition/fuel/fuel.gazebosim.org/openrobotics/models/malevisitoronphone/1/meshes/MaleVisitorStatic_Col.obj]
[ruby $(which ign) gazebo-1] [Dbg] [Zip.cc:180] Created file [/local/marius/.ignition/fuel/fuel.gazebosim.org/openrobotics/models/malevisitoronphone/1/meshes/Phone.png]
[ruby $(which ign) gazebo-1] [Dbg] [Zip.cc:180] Created file [/local/marius/.ignition/fuel/fuel.gazebosim.org/openrobotics/models/malevisitoronphone/1/model.config]
[ruby $(which ign) gazebo-1] [Dbg] [Zip.cc:180] Created file [/local/marius/.ignition/fuel/fuel.gazebosim.org/openrobotics/models/malevisitoronphone/1/model.sdf]
[ruby $(which ign) gazebo-1] [Dbg] [Zip.cc:180] Created file [/local/marius/.ignition/fuel/fuel.gazebosim.org/openrobotics/models/malevisitoronphone/1/thumbnails/1.png]
[ruby $(which ign) gazebo-1] [Dbg] [Zip.cc:180] Created file [/local/marius/.ignition/fuel/fuel.gazebosim.org/openrobotics/models/malevisitoronphone/1/thumbnails/2.png]
[ruby $(which ign) gazebo-1] [Dbg] [Zip.cc:180] Created file [/local/marius/.ignition/fuel/fuel.gazebosim.org/openrobotics/models/malevisitoronphone/1/thumbnails/3.png]
[ruby $(which ign) gazebo-1] [Dbg] [Zip.cc:180] Created file [/local/marius/.ignition/fuel/fuel.gazebosim.org/openrobotics/models/malevisitoronphone/1/thumbnails/4.png]
[ruby $(which ign) gazebo-1] [Dbg] [Zip.cc:180] Created file [/local/marius/.ignition/fuel/fuel.gazebosim.org/openrobotics/models/malevisitoronphone/1/thumbnails/5.png]
[create-8] [INFO] [1694077398.012714402] [ros_gz_sim]: Requesting list of world names.
[create-7] [INFO] [1694077398.018412506] [ros_gz_sim]: Requesting list of world names.
[ruby $(which ign) gazebo-1] [Dbg] [LocalCache.cc:601] Model [malevisitoronphone] loading resource from another model, named [MaleVisitorOnPhone]. On Blueprint (ign-fuel-tools 3) and Citadel (ign-fuel-tools 4), [MaleVisitorOnPhone] is ignored. From Dome (ign-fuel-tools 5), [malevisitoronphone] will be used. If [MaleVisitorOnPhone] is not a model belonging to owner [openrobotics], fix your SDF file!
[ruby $(which ign) gazebo-1] [Dbg] [LocalCache.cc:601] Model [malevisitoronphone] loading resource from another model, named [MaleVisitorOnPhone]. On Blueprint (ign-fuel-tools 3) and Citadel (ign-fuel-tools 4), [MaleVisitorOnPhone] is ignored. From Dome (ign-fuel-tools 5), [malevisitoronphone] will be used. If [MaleVisitorOnPhone] is not a model belonging to owner [openrobotics], fix your SDF file!
[ruby $(which ign) gazebo-1] [Msg] Downloading model [https://fuel.gazebosim.org/plateau/models/casual female]
[ruby $(which ign) gazebo-1] [Dbg] [FuelClient.cc:2002] Downloading from a referral link [https://gz-fuel-resources-production.s3.amazonaws.com/plateau/89535190-0228-4f69-9f9c-ab9f58034d12/.zips/1.zip?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAS5OHIRKDEEBENL5G%2F20230907%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20230907T090316Z&X-Amz-Expires=3600&X-Amz-SignedHeaders=host&X-Amz-Signature=3e3913f7858e8cbde6bb831dbeb4cc2a79eee8de7888bc12fab9b896cae179a6]
[... cut ...]
[create-8] [INFO] [1694077413.014311996] [ros_gz_sim]: Requesting list of world names.
[create-7] [INFO] [1694077413.019657932] [ros_gz_sim]: Requesting list of world names.
[create-8] [INFO] [1694077418.015110473] [ros_gz_sim]: Requesting list of world names.
[create-7] [INFO] [1694077418.020120229] [ros_gz_sim]: Requesting list of world names.
```
There is just a black screen that closed without any interaction when this `[INFO] [ruby $(which ign) gazebo-1]: process has finished cleanly [pid 129365]` was logged

hilary-luo, 2023-09-13:
@lizsterkenburg @teufelweich Can you clarify details about your environment? 
- OS version
- ROS Distro
- is it being run natively or in a VM or in a container or WSL?
- and how are you launching the sim?

teufelweich, 2023-09-19:
> @lizsterkenburg @teufelweich Can you clarify details about your environment?
> 
>     * OS version
Linux Mint 21.1 (Kernel 5.15.0-83-generic)
> 
>     * ROS Distro
Humble
> 
>     * is it being run natively or in a VM or in a container or WSL?
Running natively
> 
>     * and how are you launching the sim?
`ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py`

I updated some ROS packages. Now the simulator starts. I can see the simulated world and HMI GUI, but no robot is spawned inside the world.

teufelweich, 2023-09-25:
Yes, the terminal output has changed. This is the complete output:

```
ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py
[INFO] [launch]: All log files can be found below /local/marius/.ros/log/2023-09-25-15-28-00-434348-lenbook8a-209006
[INFO] [launch]: Default logging verbosity is set to INFO
ign_args is deprecated, migrate to gz_args!
[INFO] [ruby $(which ign) gazebo-1]: process started with pid [209009]
[INFO] [parameter_bridge-2]: process started with pid [209011]
[INFO] [robot_state_publisher-3]: process started with pid [209014]
[INFO] [joint_state_publisher-4]: process started with pid [209017]
[INFO] [robot_state_publisher-5]: process started with pid [209019]
[INFO] [static_transform_publisher-6]: process started with pid [209021]
[INFO] [create-7]: process started with pid [209023]
[INFO] [create-8]: process started with pid [209025]
[INFO] [parameter_bridge-9]: process started with pid [209027]
[INFO] [parameter_bridge-10]: process started with pid [209029]
[INFO] [parameter_bridge-11]: process started with pid [209033]
[INFO] [parameter_bridge-12]: process started with pid [209037]
[INFO] [parameter_bridge-13]: process started with pid [209039]
[INFO] [parameter_bridge-14]: process started with pid [209041]
[INFO] [parameter_bridge-15]: process started with pid [209043]
[INFO] [parameter_bridge-16]: process started with pid [209045]
[INFO] [parameter_bridge-17]: process started with pid [209047]
[INFO] [parameter_bridge-18]: process started with pid [209049]
[INFO] [parameter_bridge-19]: process started with pid [209051]
[INFO] [parameter_bridge-20]: process started with pid [209053]
[INFO] [parameter_bridge-21]: process started with pid [209055]
[INFO] [parameter_bridge-22]: process started with pid [209057]
[INFO] [parameter_bridge-23]: process started with pid [209059]
[INFO] [parameter_bridge-24]: process started with pid [209061]
[INFO] [parameter_bridge-25]: process started with pid [209063]
[INFO] [parameter_bridge-26]: process started with pid [209065]
[INFO] [parameter_bridge-27]: process started with pid [209067]
[INFO] [parameter_bridge-28]: process started with pid [209069]
[INFO] [parameter_bridge-29]: process started with pid [209071]
[INFO] [turtlebot4_node-30]: process started with pid [209073]
[INFO] [turtlebot4_ignition_hmi_node-31]: process started with pid [209075]
[INFO] [spawner-32]: process started with pid [209077]
[INFO] [hazards_vector_publisher-33]: process started with pid [209079]
[INFO] [ir_intensity_vector_publisher-34]: process started with pid [209086]
[INFO] [motion_control-35]: process started with pid [209091]
[INFO] [wheel_status_publisher-36]: process started with pid [209118]
[INFO] [mock_publisher-37]: process started with pid [209151]
[INFO] [robot_state-38]: process started with pid [209174]
[INFO] [kidnap_estimator_publisher-39]: process started with pid [209186]
[INFO] [ui_mgr-40]: process started with pid [209221]
[INFO] [pose_republisher_node-41]: process started with pid [209243]
[INFO] [sensors_node-42]: process started with pid [209249]
[INFO] [interface_buttons_node-43]: process started with pid [209297]
[INFO] [static_transform_publisher-44]: process started with pid [209335]
[INFO] [static_transform_publisher-45]: process started with pid [209424]
[parameter_bridge-2] [INFO] [1695648483.528835954] [clock_bridge]: Creating GZ->ROS Bridge: [/clock (ignition.msgs.Clock) -> /clock (rosgraph_msgs/msg/Clock)] (Lazy 0)
[robot_state_publisher-3] [WARN] [1695648483.363264969] [kdl_parser]: The root link base_link has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
[robot_state_publisher-3] [INFO] [1695648483.365760272] [robot_state_publisher]: got segment base_link
[robot_state_publisher-3] [INFO] [1695648483.365897481] [robot_state_publisher]: got segment bump_front_center
[robot_state_publisher-3] [INFO] [1695648483.365915015] [robot_state_publisher]: got segment bump_front_left
[robot_state_publisher-3] [INFO] [1695648483.365927628] [robot_state_publisher]: got segment bump_front_right
[robot_state_publisher-3] [INFO] [1695648483.365940002] [robot_state_publisher]: got segment bump_left
[robot_state_publisher-3] [INFO] [1695648483.365951574] [robot_state_publisher]: got segment bump_right
[robot_state_publisher-3] [INFO] [1695648483.365963506] [robot_state_publisher]: got segment bumper
[robot_state_publisher-3] [INFO] [1695648483.365975088] [robot_state_publisher]: got segment button_1
[robot_state_publisher-3] [INFO] [1695648483.365986740] [robot_state_publisher]: got segment button_2
[robot_state_publisher-3] [INFO] [1695648483.365998042] [robot_state_publisher]: got segment button_power
[robot_state_publisher-3] [INFO] [1695648483.366009994] [robot_state_publisher]: got segment cliff_front_left
[robot_state_publisher-3] [INFO] [1695648483.366021837] [robot_state_publisher]: got segment cliff_front_right
[robot_state_publisher-3] [INFO] [1695648483.366033870] [robot_state_publisher]: got segment cliff_side_left
[robot_state_publisher-3] [INFO] [1695648483.366045451] [robot_state_publisher]: got segment cliff_side_right
[robot_state_publisher-3] [INFO] [1695648483.366056883] [robot_state_publisher]: got segment front_caster_link
[robot_state_publisher-3] [INFO] [1695648483.366068695] [robot_state_publisher]: got segment front_left_bottom_weight_block
[... 367 lines cut ...]
[... cut ...]
[create-7] [INFO] [1695648509.826881993] [ros_gz_sim]: Waiting messages on topic [robot_description].
[create-8] [INFO] [1695648510.026921144] [ros_gz_sim]: Waiting messages on topic [standard_dock_description].
[create-7] [INFO] [1695648510.827087769] [ros_gz_sim]: Waiting messages on topic [robot_description].
[create-8] [INFO] [1695648511.027074051] [ros_gz_sim]: Waiting messages on topic [standard_dock_description].
```
There seems to be an issue with model deserialization. But they are just warnings

lizsterkenburg, 2023-09-26:
@hilary-luo   
The details of my environment: 

    * OS version
Ubuntu 22.04
    * ROS Distro
ROS2 Humble
    * is it being run natively or in a VM or in a container or WSL?
Natively
    * and how are you launching the sim?
ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py

Side note, I can now start the warehouse world with the robot, and I get the error for requesting world names if I use my own world file. So my issue might be a different issue.

hilary-luo, 2023-10-12:
Thanks for your patience, things have been busy for me. 

@lizsterkenburg it does sound like you have a different issue with your own world file. 

@teufelweich Are you still having this issue? If so, have you installed it from source or with debians? This sort of error could be caused by missing dependencies or some packages not aligning:

```
[ruby $(which ign) gazebo-1] ign gazebo server: symbol lookup error: /usr/lib/x86_64-linux-gnu/ign-gazebo-6/plugins/libignition-gazebo-physics-system.so: undefined symbol: _ZN8ignition6gazebo2v610components7Factory8InstanceEv
```
Try running `sudo apt update` and `sudo apt install` to make sure all of your packages are up to date and if you are running from source then verify the rosdeps.

Have you made any changes to the default warehouse or turtlebot models?

hilary-luo, 2023-11-16:
This issue is being closed due to inactivity. If you are still experiencing the issue, feel free to reopen this ticket when you are ready to continue the troubleshooting process.

closed 2023-11-16 by hilary-luo

flxinxout, 2023-12-05:
Hello, I have exactly the same problems. the same config ubuntu 22.04 and ROS2 humble

aldodiaz-UFG, 2024-03-18:
Hi,
Have the same problem reported here (Ubuntu 22 + Docker + ROS2 Humble + Turtlebot4). Gazebo simulation window freezes on messages of downloading world.

Do you have pointer to a working image we can use out of the box perhaps?

federicociresola, 2024-03-22:
Hi,
I have the same problem on my PC
- OS version: Ubuntu 22.04
- ROS2 version: Humble
- GZ version: Garden

I noticed that the gazebo is not able to reach the server to download all resource. Does anyone resolve this problem?

federicociresola, 2024-03-22:
During the debug, I discovered that the gazebo shows these error:

```
[...]
[Wrn] [FuelClient.cc:1978] The fuel.ignitionrobotics.org URL is deprecrated. Please change https://fuel.ignitionrobotics.org to https://fuel.gazebosim.org
[...]
[ruby $(which ign) gazebo-1] [Err] [FuelClient.cc:717] Failed to download model.
[ruby $(which ign) gazebo-1]   Server: https://fuel.ignitionrobotics.org
[ruby $(which ign) gazebo-1]   Route: openrobotics/models/artifact proximity detector/tip/artifact proximity detector.zip
[ruby $(which ign) gazebo-1]   REST response code: 504
[ruby $(which ign) gazebo-1] [Err] [Server.cc:139] Error Code 13: [/sdf/model[@name="rescue_randy_sitting"]/include[0]/uri:/home/toshi/.ignition/fuel/fuel.gazebosim.org/openrobotics/models/rescue randy sitting/8/model.sdf:L6]: Msg: Unable to find uri[https://fuel.ignitionrobotics.org/1.0/OpenRobotics/models/Artifact Proximity Detector]
```
I tried to replace all file that contains https://fuel.ignitionrobotics.org with https://fuel.gazebosim.org but it didn't resolve the problem. In addition, I notice that if you change the world with maze (whose doesn't contain uri) it works without errors:
```
ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py world:=maze 
```
Therefore, I think that the problem is the gz_server is not able to reach the fuel remote repository

hilary-luo, 2024-03-27:
@federicociresola can you please try again and confirm that those errors are no longer present?

hilary-luo, 2024-03-27:
@federicociresola Thanks for the feedback. It seems like the issue only went away temporarily and is back. Please follow along [here](https://github.com/gazebo-web/fuel-server/issues/19#issuecomment-2022758015) since I'd rather not keep coming back to this closed issue. Also, in the future, it's better for you to open your own issue than to comment on a closed issue, especially when you have a different error message than the OP.

vivekvardhan-god, 2024-09-29:
Diabling the firewall using sudo ufw diable worked for me

deepak-anzo, 2024-11-18:
It worked for me as well but why?

deepak-anzo, 2024-11-18:
> Diabling the firewall using sudo ufw diable worked for me

It worked for me as well but why?

hilary-luo, 2024-11-18:
@deepak-anzo these model files are downloaded from the gazebo servers. Likely your firewall was blocking the program from accessing these servers (a common type of firewall protection).