turtlebot/turtlebot4_simulator #47: Turtlebot in custom world file in humble.

vamsi1961, 2023-07-16:
I have an sdf file. How to do navigation in rviz. How to convert to pgm, yaml. Can we do these in humble?

Any suggestions?

hilary-luo, 2023-07-18:
You can launch the [turtlebot sim](https://turtlebot.github.io/turtlebot4-user-manual/software/turtlebot4_simulator.html) with your custom world and then follow the [generate map tutorial](https://turtlebot.github.io/turtlebot4-user-manual/tutorials/generate_map.html) to map the world and then save it.

vamsi1961, 2023-07-19:
I have an sdf file suitable for ignition. But in generate a map article it is for real word. How to generate map for sdf file.

hilary-luo, 2023-07-20:
You can launch the [turtlebot sim with that sdf file using the world parameter](https://turtlebot.github.io/turtlebot4-user-manual/software/turtlebot4_simulator.html#ignition-bringup). This will allow the robot to be spawned in your custom world. You may also need to modify the x and y coordinates to change where the robot spawns. It might look like this (adjust the path and x/y for your case):
`ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py slam:=true nav2:=true rviz:=true world:=/home/administrator/custom_world_file x:=2.0 y:=5.0`

Once you have launched the sim and your robot is present, you can drive the robot around to map your world as in the [generating a map tutorial](https://turtlebot.github.io/turtlebot4-user-manual/tutorials/generate_map.html#drive-the-turtlebot-4). When you save the map using the command given there, it will save the pgm and yaml files.

vamsi1961, 2023-07-24:
Hey hilary-luo thanks for replying. 

I think I am missing some thing. In [generate a map](https://turtlebot.github.io/turtlebot4-user-manual/tutorials/generate_map.html#drive-the-turtlebot-4) how to do for custom world. Can you please guide me through this.

vamsi1961, 2023-07-24:
I launched with my sdf file. Igition is popping and unexpected closures, but it should remain stable and operational.

hilary-luo, 2023-07-24:
So the state of your system prior to doing the  generate map tutorial should be that your robot is in an environment that needs to be mapped. In gazebo (or in the real world if you were just using the robot instead of sim), you can see where the robot is and the obstacles around it, that the robot in turn can detect with its sensors. The goal of that tutorial is to generate the map files for that given environment. So for that reason, I indicated first that you need to launch the simulation with your custom world. This is the full simulation of the real world such that the robot is governed by gravity and it will collide with the obstacles if you were to try and drive it into an obstacle. You can use this simulation in the place of a real robot for any of the tutorials. 

At this point your simulation robot should be launched in the custom world. The generating a map tutorial then takes you through turning on SLAM (Simultaneous Location and Mapping) to create the map. There is no starting file required, it will be mapping from scratch. So open a new terminal and run `ros2 launch turtlebot4_navigation slam.launch.py`. SLAM will start using the lidar readings to build the map. And then open one more terminal to launch rviz `ros2 launch turtlebot4_viz view_robot.launch.py`. This launches your visulization of the mapping procedure. 

The next step is to drive your robot around. As you drive the robot around, the SLAM node will use the lidar scans to build a map. You will need to drive around the whole environment (or all sections that you want to map), letting the lidar see every edge of an obstacle that you want mapped. In simulation these lidar scans are determined in the gazebo simulation based on where the obstacles are located relative to the robot. 

You can drive the robot around using the teleop panel on the right side of the gazebo window or using the terminal based [keyboard teleoperation](https://turtlebot.github.io/turtlebot4-user-manual/tutorials/driving.html#keyboard-teleoperation). When you are done you can then move onto the [save the map](https://turtlebot.github.io/turtlebot4-user-manual/tutorials/generate_map.html#save-the-map) step. This saves the map to your current directory and can be loaded into the system in the future. 

Does that resolve your confusion?

If you are getting errors in running the sim and are looking for support then please provide the full terminal outputs.

vamsi1961, 2023-07-25:
Thanks for your time. I appreciate your help.
I am using humble. 
My world is in Ignition.
Can we generate map for ignition world using above method.

vamsi1961, 2023-07-25:
ARGUMENTS = [
    DeclareLaunchArgument('namespace', default_value='',
                          description='Robot namespace'),
    DeclareLaunchArgument('rviz', default_value='false',
                          choices=['true', 'false'], description='Start rviz.'),
    DeclareLaunchArgument('world', default_value='maze',
                          description='Ignition World'),
    DeclareLaunchArgument('model', default_value='standard',
                          choices=['standard', 'lite'],
                          description='Turtlebot4 Model'),
]

In the above argument of turtlebot4_ignition.launch.py my world is in other directory. How to import it. I am facing many errors while importing world from another package?

hilary-luo, 2023-07-26:
HI @vamsi1961 

Responding to the collective last few messages:
Yes you can do what I am describing with humble and gazebo ignition. All of the information I have shared is for humble and gazebo ignition. I know the terminology can be confusing because Ignition was renamed to Gazebo and old Gazebo has been renamed to Gazebo Classic. 

In the statement about launching the sim I was referring to my previous message giving the sim launch command. 

You have shared contents of the launch file but you should not be editing that file. You should be doing everything by calling the launch files from the terminal and passing through the parameters.

In an attempt to clarify, here are step by step instructions for running it from source. If one of the steps below does not work then you need to share exactly the command that you ran and the full terminal output. Unfortunately, the little bit of information you have given about being unable to launch is not enough information for anyone to help. 

1. Clone the turtlebot4_simulator Put your sdf file in the worlds folder
2. Build the package (`colcon build`)
3. Source the package in the terminal (`source install/setup.bash`)
4. Run `ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py slam:=true nav2:=true rviz:=true world:=custom_world_file` but replace custom_world_file with the name of your sdf world file.
5. Press play in Gazebo

This should get you as far as having the sim up and running and the mapping process started. Again, if any of these steps does not work then please provide the steps you have taken so far, the exact command that failed and the full terminal output of the failure.

vamsi1961, 2023-08-06:
ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py
slam:=true nav2:=true rviz:=true
world:=/home/administrator/custom_world_file x:=2.0 y:=5.0

Unable to access custom world from other location

hilary-luo, 2023-08-08:
HI @vamsi1961,

After looking into it further, you do need to build from source and place the sdf file in the worlds folder. I will ensure that the documentation gets updated to reflect this. Please see my instructions for running it from source in my last message.

closed 2023-09-11 by hilary-luo