turtlebot/turtlebot4_simulator #49: IMU in simulation

y2d2, 2023-07-21:
**Describe the feature you would like**
Maybe I might have missed it, but it seems the IMU is not simulated. I am not sure whether TB4 in general has an IMU of its own, but I know the OAKD-pro has one and it would be lovely to have the IMU simulated so VIO (like VINS etc) can be used as well in simulation.  

I apologize if this is not the way to ask these things.

y2d2, 2023-07-25:
Thanks ! Looking forward to the update.

closed 2023-07-25 by y2d2

svdeepak99, 2024-05-28:
> The base Create 3 does have an IMU, and it does get [simulated](https://github.com/iRobotEducation/create3_sim/blob/7fccefe91a300ea945aa81df09e7db37898b4eda/irobot_create_common/irobot_create_description/urdf/create3.urdf.xacro#L116). The missing piece is the gz_to_ros bridge for the imu topic. We will make a PR for that on the `create3_sim` repo to add this.

@roni-kreinin any updates on this?