turtlebot/turtlebot4_simulator #51: Gazebo failing to launch with Turtlebot 4 simulation packages

Cam-uzi, 2023-08-30:
**Please provide the following information:**
 - OS: Ubuntu Desktop 22.04
 - ROS Distro: ROS 2 Humble
 - Installed metapackage through apt:
 - Package version: Debian package, sudo apt install ros-humble-turtlebot4-simulator

After successfully installing gazebo using a WLS 2 system on Windows 10 with Ubuntu Desktop 22.04 with Ros 2 humble, I attempted to launch gazebo using the command ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py, however, gazebo is unable to launch and the following error continuously prints in the command screen.  

[image]

The command seems to be fetching a website that no longer exists. The updated website for Gazebo models is https://app.gazebosim.org/fuel/worlds. I am attempting to solve this issue myself by manually changing the website URL in the code however I am currently unsuccessful. Do you have any advice to fix this issue? Are there any plans to fix this error as many of my colleagues have this issue? If you have any help to offer, please respond as soon as possible.

hilary-luo, 2023-08-30:
Hi @Cam-uzi, 

Although that link is outdated, it is still functional, I just tested it to make sure. Additionally, as a note for you, the path used to download the files is not the same url path as the website that you would visit from your browse.

Can you confirm that you have internet access from WSL2? (`ping 8.8.8.8` to ensure that you can connect to the internet and then `ping google.com` to verify that it is able to resolve the address).

Cam-uzi, 2023-08-30:
Good Afternoon, 

I can confirm that I have internet access from WSL2.

hilary-luo, 2023-08-31:
I am unable to replicate this issue on Ubuntu. Unfortunately we don't work with WSL and this seems to be a WSL + Gazebo issue, not specific to the turtlebot packages.

If you have already tried updating `https://fuel.ignitionrobotics.org` to the updated path `https://fuel.gazebosim.org` with no success, I would suggest opening a question on https://robotics.stackexchange.com/. I have heard of there being firewall issues when using WSL so maybe that is a direction to pursue.

closed 2023-09-05 by Cam-uzi

alexdo105, 2024-04-10:
Hello, I am having trouble when trying to run the turtlebot4 simulator packages, I have installed the turtlebot4 humble gazebo simulator packages as shown in the manual:
sudo apt install ros-dev-tools
sudo apt-get update && sudo apt-get install wget
sudo sh -c 'echo "deb http://packages.osrfoundation.org/gazebo/ubuntu-stable `lsb_release -cs` main" > /etc/apt/sources.list.d/gazebo-stable.list'
wget http://packages.osrfoundation.org/gazebo.key -O - | sudo apt-key add -
sudo apt-get update && sudo apt-get install ignition-fortress
sudo apt update
sudo apt install ros-humble-turtlebot4-simulator

When launching the simulator gazebo just opens and stay in a black screen, I launched it as the manual shows:
ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py

I am running everything natively, my computer runs linux.
I hope you can help because this project is very important for me.

automessage, 2024-04-10:
Hello!
Im running into this same issue on Ubuntu 22.04 running natively. I installed the packages following the instructions in the documentation, but when i try to start the simulation, it can't fetch models from the URI. The only difference to the original posters problem is the response code, which in my case is 504.

My error output:

```
[ruby $(which ign) gazebo-1] [Err] [FuelClient.cc:717] Failed to download model.
[ruby $(which ign) gazebo-1]   Server: https://fuel.ignitionrobotics.org
[ruby $(which ign) gazebo-1]   Route: movai/models/shelf_big/tip/shelf_big.zip
[ruby $(which ign) gazebo-1]   REST response code: 504
[ruby $(which ign) gazebo-1] [Wrn] [FuelClient.cc:1978] The fuel.ignitionrobotics.org URL is deprecrated. Pleasse change https://fuel.ignitionrobotics.org to https://fuel.gazebosim.org 
``` 
If i change the domain in the URI in the source file (warehouse.sdf) to fuel.gazebosim.org and build again, but i didn't solve the problem.
Is there something else i could do to make the simulator work?

hilary-luo, 2024-04-10:
@alexdo105 I have responded to you on a different ticket.

@automessage This is an issue with the gazebo models right now. See the current work arounds here: https://github.com/clearpathrobotics/clearpath_simulator/issues/30#issuecomment-2020693160 (uses the same warehouse world). Or for the time being launch the maze world only. Also, if your sim is currently working then be warned that there is a current bug with the latest version of some of the packages so I would suggest not updating your packages until after [this bug](https://github.com/turtlebot/turtlebot4_simulator/issues/72#issuecomment-2047704021) is closed.

automessage, 2024-04-10:
@hilary-luo thank you for your response, i will look into that!