turtlebot/turtlebot4_simulator #54: Lidar range in ign gazebo is small affecting mapping and localization

harishkumarbalaji, 2023-10-13:
**Please provide the following information:**
 - OS: Ubuntu 22 and also in Ubuntu 20 isaac_ros_container(source build)
 - ROS Distro: humble and galactic
 - Built from source or installed: Both
 - Package version: https://github.com/ros2-gbp/turtlebot4_simulator-release.git - release/humble/turtlebot4_simulator/1.0.0-1

 **Expected behavior**
Lidar rays should have a max range of 12 meters (as per RpLidar A1 specs). 

 **Actual behavior**
Lidar rays have a max range of less than 1 meter. This affects localization and mapping.
[image]

**To Reproduce**
Provide the steps to reproduce:
1. launch ignition bring up 
```
ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py slam:=false nav2:=false rviz:=false
```
2. Click on **visualize LIDAR** option in the ign gazebo GUI
3. Select the last topic after refreshing the topic list.

**Other notes**
This if found in all the places where I run. Like,
1. Lapton with and without GPU
2. source build and binary

hilary-luo, 2023-10-26:
Can you verify that the GPU is actively been used by the simulation? (usually can be checked using nvidia-smi and then ensuring the simulation process is present)

hilary-luo, 2023-11-20:
This issue is being closed due to inactivity. If you are still experiencing the issue, feel free to reopen this ticket when you are ready to continue the troubleshooting process.

closed 2023-11-20 by hilary-luo

harishkumarbalaji, 2024-10-23:
I am facing the same issue, but more weird that only a semi-circle is seen when we use the Visualise LIDAR plugin.
[image]

Nvidia-smi output

```hkb@hkb:~/turtlebot4_ws$ nvidia-smi
Wed Oct 23 17:41:59 2024       
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 550.107.02             Driver Version: 550.107.02     CUDA Version: 12.4     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA GeForce GTX 1660 ...    Off |   00000000:01:00.0 Off |                  N/A |
| N/A   53C    P0             18W /   60W |       6MiB /   6144MiB |      0%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
                                                                                         
+-----------------------------------------------------------------------------------------+
| Processes:                                                                              |
|  GPU   GI   CI        PID   Type   Process name                              GPU Memory |
|        ID   ID                                                               Usage      |
|=========================================================================================|
|    0   N/A  N/A      2058      G   /usr/lib/xorg/Xorg                              4MiB |
+-----------------------------------------------------------------------------------------+```