turtlebot/turtlebot4_simulator #56: Timed out waiting for transform from base_link to odom to become available

harshbhatt99, 2023-10-30:
- OS: Ubuntu 22.04
 - ROS Distro: ROS2 Humble
 - Built from source or installed: Built from source
 - Package version: git clone [https://github.com/turtlebot/turtlebot4.git -b humble](url), [git clone https://github.com/turtlebot/turtlebot4_simulator.git -b humble](url) (Following:(https://turtlebot.github.io/turtlebot4-user-manual/software/turtlebot4_common.html))

 **Expected behaviour**
RViz should load with all static transforms with warehouse map and "2D Pose Estimate" and "Nav2 Goal" should work in RViz 

 **Actual behaviour**
RViz loading with RobotModel error and FixedFrame error (map does not exist)

**To Reproduce**
Provide the steps to reproduce:
1. Run `ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py nav2:=true slam:=false localization:=true rviz:=true`

**Errors**
1. RViz > RobotModel > Status: Error > base_link > No transform from [base_link] to [map] 
2. Same error in RViz for all other links.
3. Terminal error as follows: 
`[controller_server-49] [INFO] [1698695914.949777800] [local_costmap.local_costmap]: Timed out waiting for transform from base_link to odom to become available, tf error: Could not find a connection between 'odom' and 'base_link' because they are not part of the same tree.Tf has two or more unconnected trees.`

**Other notes**
I believe by running the mentioned launch.py file, it should automatically load the warehouse.yaml map. It is loading that map in RViz but the mentioned error is also there and 2D Pose Estimate and Nav2 Goal is not working in RViz.

hilary-luo, 2023-10-30:
Have you pressed play on the gazebo simulation (if you have in the bottom left it should now display a pause button) and then set your initial 2d pose estimate in rviz?

harshbhatt99, 2023-10-30:
Got it. Didn't know of this step. Thank you.

closed 2023-11-10 by hilary-luo