turtlebot/turtlebot4_simulator #59: stuck at Requesting list of world names

abdosalem490, 2023-11-10:
- OS: Ubuntu 22.04
 - ROS Distro: Humble
 - Built from source or installed: source
 - Package version:  [e21bd2c](https://github.com/turtlebot/turtlebot4_simulator/commit/e21bd2c565b6a220c2f1156082d17855c9bc4dc3)

 **Expected behaviour**
 ignition gazebo shall run and open the simulator.

 **Actual behaviour**
 controller manager is not available and is stuck Requesting list of world names.

**To Reproduce**
Provide the steps to reproduce:
1. `ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py`
2. Reuslt:
```
[INFO] [launch]: All log files can be found below /home/abdosalm/.ros/log/2023-11-10-22-37-52-587707-abdosalm-Alienware-m17-R3-47184
[INFO] [launch]: Default logging verbosity is set to INFO
ign_args is deprecated, migrate to gz_args!
[INFO] [ruby $(which ign) gazebo-1]: process started with pid [47187]
[INFO] [parameter_bridge-2]: process started with pid [47190]
[INFO] [robot_state_publisher-3]: process started with pid [47192]
[INFO] [joint_state_publisher-4]: process started with pid [47195]
[INFO] [robot_state_publisher-5]: process started with pid [47197]
[INFO] [static_transform_publisher-6]: process started with pid [47199]
[INFO] [create-7]: process started with pid [47201]
[INFO] [create-8]: process started with pid [47203]
[INFO] [parameter_bridge-9]: process started with pid [47205]
[INFO] [parameter_bridge-10]: process started with pid [47209]
[INFO] [parameter_bridge-11]: process started with pid [47212]
[INFO] [parameter_bridge-12]: process started with pid [47215]
[INFO] [parameter_bridge-13]: process started with pid [47217]
[INFO] [parameter_bridge-14]: process started with pid [47219]
[INFO] [parameter_bridge-15]: process started with pid [47221]
[INFO] [parameter_bridge-16]: process started with pid [47229]
[INFO] [parameter_bridge-17]: process started with pid [47253]
[INFO] [parameter_bridge-18]: process started with pid [47256]
[INFO] [parameter_bridge-19]: process started with pid [47258]
[INFO] [parameter_bridge-20]: process started with pid [47277]
[INFO] [parameter_bridge-21]: process started with pid [47292]
[INFO] [parameter_bridge-22]: process started with pid [47334]
[INFO] [parameter_bridge-23]: process started with pid [47375]
[INFO] [parameter_bridge-24]: process started with pid [47379]
[INFO] [parameter_bridge-25]: process started with pid [47433]
[INFO] [parameter_bridge-26]: process started with pid [47450]
[INFO] [parameter_bridge-27]: process started with pid [47461]
[INFO] [parameter_bridge-28]: process started with pid [47486]
[INFO] [parameter_bridge-29]: process started with pid [47502]
[INFO] [turtlebot4_node-30]: process started with pid [47508]
[INFO] [turtlebot4_ignition_hmi_node-31]: process started with pid [47519]
[INFO] [spawner-32]: process started with pid [47571]
[INFO] [hazards_vector_publisher-33]: process started with pid [47591]
[INFO] [ir_intensity_vector_publisher-34]: process started with pid [47618]
[INFO] [motion_control-35]: process started with pid [47620]
[INFO] [wheel_status_publisher-36]: process started with pid [47622]
[INFO] [mock_publisher-37]: process started with pid [47624]
[INFO] [robot_state-38]: process started with pid [47626]
[INFO] [kidnap_estimator_publisher-39]: process started with pid [47639]
[INFO] [ui_mgr-40]: process started with pid [47651]
[INFO] [pose_republisher_node-41]: process started with pid [47653]
[INFO] [sensors_node-42]: process started with pid [47655]
[INFO] [interface_buttons_node-43]: process started with pid [47657]
[INFO] [static_transform_publisher-44]: process started with pid [47659]
[INFO] [static_transform_publisher-45]: process started with pid [47686]
[parameter_bridge-2] [INFO] [1699648676.439669965] [clock_bridge]: Creating GZ->ROS Bridge: [/clock (ignition.msgs.Clock) -> /clock (rosgraph_msgs/msg/Clock)] (Lazy 0)
[robot_state_publisher-3] [WARN] [1699648676.330157552] [kdl_parser]: The root link base_link has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
[robot_state_publisher-3] [INFO] [1699648676.330368347] [robot_state_publisher]: got segment base_link
[robot_state_publisher-3] [INFO] [1699648676.330511260] [robot_state_publisher]: got segment bump_front_center
[robot_state_publisher-3] [INFO] [1699648676.330527167] [robot_state_publisher]: got segment bump_front_left
[robot_state_publisher-3] [INFO] [1699648676.330535320] [robot_state_publisher]: got segment bump_front_right
[... 246 lines cut ...]
[create-8] [INFO] [1699648891.376363220] [ros_gz_sim]: Requesting list of world names.
[create-7] [INFO] [1699648891.376385571] [ros_gz_sim]: Requesting list of world names.
[create-8] [INFO] [1699648896.376998846] [ros_gz_sim]: Requesting list of world names.
[create-7] [INFO] [1699648896.377031935] [ros_gz_sim]: Requesting list of world names.
[create-8] [INFO] [1699648901.377554592] [ros_gz_sim]: Requesting list of world names.
[create-7] [INFO] [1699648901.377582062] [ros_gz_sim]: Requesting list of world names.
[create-8] [INFO] [1699648906.378193856] [ros_gz_sim]: Requesting list of world names.
[create-7] [INFO] [1699648906.378214520] [ros_gz_sim]: Requesting list of world names.
[create-8] [INFO] [1699648911.378813648] [ros_gz_sim]: Requesting list of world names.
[create-7] [INFO] [1699648911.378851439] [ros_gz_sim]: Requesting list of world names.
[create-7] [INFO] [1699648916.379436227] [ros_gz_sim]: Requesting list of world names.
[create-8] [INFO] [1699648916.379514368] [ros_gz_sim]: Requesting list of world names.
[create-8] [INFO] [1699648921.380059016] [ros_gz_sim]: Requesting list of world names.
[create-7] [INFO] [1699648921.380106151] [ros_gz_sim]: Requesting list of world names.
[create-7] [INFO] [1699648926.380653874] [ros_gz_sim]: Requesting list of world names.
[create-8] [INFO] [1699648926.380685990] [ros_gz_sim]: Requesting list of world names.
[create-8] [INFO] [1699648931.380987574] [ros_gz_sim]: Requesting list of world names.
[create-7] [INFO] [1699648931.380996178] [ros_gz_sim]: Requesting list of world names.
[create-8] [INFO] [1699648936.381721703] [ros_gz_sim]: Requesting list of world names.
[create-7] [INFO] [1699648936.381782651] [ros_gz_sim]: Requesting list of world names.
[create-8] [INFO] [1699648941.382310738] [ros_gz_sim]: Requesting list of world names.
[create-7] [INFO] [1699648941.382353134] [ros_gz_sim]: Requesting list of world names.
[create-8] [INFO] [1699648946.382965001] [ros_gz_sim]: Requesting list of world names.
[create-7] [INFO] [1699648946.383059057] [ros_gz_sim]: Requesting list of world names.
```

abdosalem490, 2023-11-11:
so, I managed to solve this problem, it was because I had missing packages not installed from [the common packages](https://turtlebot.github.io/turtlebot4-user-manual/software/turtlebot4_common.html), once I installed all the common packages not only the simulation package, I managed to to run the command `ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py` without any errors.

so yes, it was my fault. now I am discovering how to modify the launch files so that I can color the robot, and change the world file to be my own custom world.

sorry for the disruption.

closed 2023-11-11 by abdosalem490

JHPHELAN, 2024-04-11:
@abdosalem490 
-Simulation says is installs all the same packages as -Common.  Installing -common & -simulation gives underlay/overlay warnings.  How, exactly, did you install the separate packages?  Other online comments say problem is fixed in source. I've installed from Debian and from source and continue to get same error.  Appreciate your insight.

hilary-luo, 2024-04-11:
@JHPHELAN There are 2 separate upstream issues currently preventing simulations from working properly for new downloads / installs. I would suggest checking these tickets to confirm that you are seeing one of the known issues. They have work arounds listed and there should be updates resolving the issues relatively soon. If your issue does not match those then feel free to open your own ticket.

1) https://github.com/turtlebot/turtlebot4_simulator/issues/72#issuecomment-2047704021
2) https://github.com/clearpathrobotics/clearpath_simulator/issues/30