turtlebot/turtlebot4_simulator #61: spaceros

yjydiancom, 2023-12-11:
**Describe the the feature you would like**
Hopefully it will build and run under spaceros

**Other notes**
I build under spaceros and three of the four packages build successfully, but the turtlebot4_ignition_toolbox package gave an error

/home/spaceros-user/turtulbot_ws# colcon build --symlink-install --packages-select turtlebot4_ignition_toolbox
[1.771s] WARNING:colcon.colcon_core.package_selection:Some selected packages are already built in one or more underlay workspaces:
        'turtlebot4_ignition_toolbox' is in: /opt/ros/humble
If a package in a merged underlay workspace is overridden and it installs headers, then all packages in the overlay must sort their include directories by workspace order. Failure to do so may result in build failures or undefined behavior at run time.
If the overridden package is used by another package in any underlay, then the overriding package in the overlay must be API and ABI compatible or undefined behavior at run time may occur.

If you understand the risks and want to override a package anyways, add the following to the command line:
        --allow-overriding turtlebot4_ignition_toolbox

This may be promoted to an error in a future release of colcon-override-check.
Starting >>> turtlebot4_ignition_toolbox
--- stderr: turtlebot4_ignition_toolbox                         
CMake Error at /home/spaceros-user/spaceros/install/rmw_implementation/share/rmw_implementation/cmake/rmw_implementation-extras.cmake:25 (message):
  The RMW implementation has been specified as 'rmw_fastrtps_cpp' via
  get_default_rmw_implementation, but rmw_implementation was built only with
  support for 'rmw_cyclonedds_cpp'.
Call Stack (most recent call first):
  /home/spaceros-user/spaceros/install/rmw_implementation/share/rmw_implementation/cmake/rmw_implementationConfig.cmake:41 (include)
  /home/spaceros-user/spaceros/install/rcl/share/rcl/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /home/spaceros-user/spaceros/install/rcl/share/rcl/cmake/rclConfig.cmake:41 (include)
  /home/spaceros-user/spaceros/install/libstatistics_collector/share/libstatistics_collector/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /home/spaceros-user/spaceros/install/libstatistics_collector/share/libstatistics_collector/cmake/libstatistics_collectorConfig.cmake:41 (include)
  /home/spaceros-user/spaceros/install/rclcpp/share/rclcpp/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /home/spaceros-user/spaceros/install/rclcpp/share/rclcpp/cmake/rclcppConfig.cmake:41 (include)
  CMakeLists.txt:20 (find_package)

yjydiancom, 2023-12-11:
[image]

yjydiancom, 2023-12-11:
[image]

hilary-luo, 2023-12-22:
The error messages seems to suggest that you need to switch the default rmw implementation to cyclonedds in order for this to build successfully in this case