turtlebot/turtlebot4_simulator #63: Cannot create a map from the custom SDF file.

harshbhatt99, 2023-12-31:
I am trying to create a map with custom SDF file but, when I launch the following commands, RViz doesn't show up the map generation and LIDAR data. SDF is loading correctly in Gazebo.

I have already added the SDF file name in the turtlebot_ignition.launch.py file.
In one terminal, I am ruuning

`ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py slam:=true nav2:=true rviz:=true`

In second terminal, I am running

`ros2 launch turtlebot4_navigation slam.launch.py`

I am following the tutorial [https://turtlebot.github.io/turtlebot4-user-manual/tutorials/generate_map.html](url)
This issue is already discussed partially in #47 and #54 but I could not find the concluding answer.

On the second terminal, I am getting this message.

`[sync_slam_toolbox_node-1] [INFO] [1703994353.087267383] [slam_toolbox]: Message Filter dropping message: frame 'turtlebot4/rplidar_link/rplidar' at time 9.075 for reason 'the timestamp on the message is earlier than all the data in the transform cache'`

I tried increasing the buffer size in slam.yaml but it doesn't seem to be working.

Here are the screenshots of the robot in Gazebo and RViz.
[image]
[image]

And here are the screenshots of the terminal messages

[image]
[image]

I tried to find the solutions from other issues and threads. I doubt if this has something related to the computational power of my system. Can you take a look and see if I am doing anything wrong or it's a system problem?

harshbhatt99, 2024-01-25:
@hilary-luo Yes, I am still experiencing this issue.

hilary-luo, 2024-01-29:
@harshbhatt99 the first thing that jumps out at me is that you seem to be launching slam twice. This would cause unpredictable results and would be the first thing to resolve. When you run: `ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py slam:=true nav2:=true rviz:=true` you are launching the simulation as well as slam, rviz and nav2. The generate map tutorial is set up for use with a physical robot, that is why it indicates for you to launch slam separately in its own window. If you want to launch slam separately then set slam to false in that command and then you can launch it separately in its own terminal window. Furthermore if you want to simplify your test then you could also set nav2 to false until you are ready to use it.

If this doesn't solve your issue, please update with the new error messages etc. It would also help if you copy and paste the output from the terminals so that we can see what it did up to the point (you can omit repeating messages, just note that they are repeated)

ajjaya18, 2024-03-04:
@harshbhatt99 Hi, you work with humble or galactic? and I have a question, I try to create a new World with ros galactic and I have problems with this, can you give me advice or can you guide me please?; @hilary-luo hi I have problems with generate a map, my turtlebot4 don't recognize the objects in the world depot, I think you know about this, the map of this world is already created in the tutorial but when I try to do the same it doesn't work, it doesn't scan anything, do I have to add some plugin to my Rplidar or something like that? please answer me.

ajjaya18, 2024-03-05:
[image]
[image]
[image]

ajjaya18, 2024-03-05:
Can you help me please?