turtlebot/turtlebot4_simulator #69: Turtlebot4 simulation

ajjaya18, 2024-03-25:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

I do not know

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

Package: ros-humble-turtlebot4-simulator
Version: 1.0.1-1jammy.20240304.161058
Priority: optional
Section: misc
Maintainer: rkreinin [rkreinin@clearpathrobotics.com](mailto:rkreinin@clearpathrobotics.com)
Installed-Size: 42.0 kB
Depends: ros-humble-turtlebot4-ignition-bringup, ros-humble-turtlebot4-ignition-gui-plugins, ros-humble-turtlebot4-ignition-toolbox, ros-humble-ros-workspace
Download-Size: 5,112 B
APT-Manual-Installed: yes
APT-Sources: http://packages.ros.org/ros2/ubuntu jammy/main amd64 Packages
Description: TODO: Package description

### Expected behaviour

I I expect to see the TurtleBot 4 simulation in Ignition Gazebo.

### Actual behaviour

I have mistakes, but I can't see the robot and also wolrd in the ignition gazebo, 

### Error messages

```bash
andrius@andrius-HP-Pavilion-Laptop-15-eh0xxx:~$ ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py
[INFO] [launch]: All log files can be found below /home/andrius/.ros/log/2024-03-25-13-58-38-320774-andrius-HP-Pavilion-Laptop-15-eh0xxx-14179
[INFO] [launch]: Default logging verbosity is set to INFO
ign_args is deprecated, migrate to gz_args!
[INFO] [ruby $(which ign) gazebo-1]: process started with pid [14182]
[INFO] [parameter_bridge-2]: process started with pid [14184]
[INFO] [robot_state_publisher-3]: process started with pid [14187]
[INFO] [joint_state_publisher-4]: process started with pid [14189]
[INFO] [robot_state_publisher-5]: process started with pid [14192]
[INFO] [static_transform_publisher-6]: process started with pid [14194]
[INFO] [create-7]: process started with pid [14204]
[INFO] [create-8]: process started with pid [14229]
[INFO] [parameter_bridge-9]: process started with pid [14232]
[INFO] [parameter_bridge-10]: process started with pid [14235]
[INFO] [parameter_bridge-11]: process started with pid [14237]
[INFO] [parameter_bridge-12]: process started with pid [14257]
[INFO] [parameter_bridge-13]: process started with pid [14264]
[INFO] [parameter_bridge-14]: process started with pid [14268]
[INFO] [parameter_bridge-15]: process started with pid [14278]
[INFO] [parameter_bridge-16]: process started with pid [14302]
[INFO] [parameter_bridge-17]: process started with pid [14311]
[INFO] [parameter_bridge-18]: process started with pid [14322]
[INFO] [parameter_bridge-19]: process started with pid [14330]
[INFO] [parameter_bridge-20]: process started with pid [14332]
[INFO] [parameter_bridge-21]: process started with pid [14334]
[... 457 lines cut ...]
[static_transform_publisher-6] [INFO] [1711393248.561700400] [rclcpp]: signal_handler(signum=2)
[robot_state_publisher-5] [INFO] [1711393248.574599583] [rclcpp]: signal_handler(signum=2)
[INFO] [parameter_bridge-9]: process has finished cleanly [pid 14232]
[robot_state_publisher-3] [INFO] [1711393248.598080358] [rclcpp]: signal_handler(signum=2)
[parameter_bridge-2] [INFO] [1711393248.610443498] [rclcpp]: signal_handler(signum=2)
[INFO] [static_transform_publisher-6]: process has finished cleanly [pid 14194]
[INFO] [robot_state_publisher-5]: process has finished cleanly [pid 14192]
[INFO] [robot_state_publisher-3]: process has finished cleanly [pid 14187]
[INFO] [parameter_bridge-2]: process has finished cleanly [pid 14184]
[INFO] [joint_state_publisher-4]: process has finished cleanly [pid 14189]
[ERROR] [parameter_bridge-28]: process has died [pid 14415, exit code -11, cmd '/opt/ros/humble/lib/ros_gz_bridge/parameter_bridge /world/warehouse/model/turtlebot4/link/rplidar_link/sensor/rplidar/scan@sensor_msgs/msg/LaserScan[ignition.msgs.LaserScan --ros-args -r __node:=lidar_bridge -r __ns:=/ --params-file /tmp/launch_params_aeuc2n66 -r /world/warehouse/model/turtlebot4/link/rplidar_link/sensor/rplidar/scan:=scan'].
[create-8] [INFO] [1711393252.474061947] [ros_gz_sim]: Requesting list of world names.
[create-7] [INFO] [1711393252.479625898] [ros_gz_sim]: Requesting list of world names.
[ERROR] [create-8]: process[create-8] failed to terminate '5' seconds after receiving 'SIGINT', escalating to 'SIGTERM'
[ERROR] [create-7]: process[create-7] failed to terminate '5' seconds after receiving 'SIGINT', escalating to 'SIGTERM'
[INFO] [create-8]: sending signal 'SIGTERM' to process[create-8]
[INFO] [create-7]: sending signal 'SIGTERM' to process[create-7]
[ERROR] [create-8]: process has died [pid 14229, exit code -15, cmd '/opt/ros/humble/lib/ros_ign_gazebo/create -name standard_dock -x 0.157 -y 0.0 -z 0.0 -Y 3.1416 -topic standard_dock_description --ros-args -r __ns:=/'].
[ERROR] [create-7]: process has died [pid 14204, exit code -15, cmd '/opt/ros/humble/lib/ros_ign_gazebo/create -name turtlebot4 -x 0.0 -y 0.0 -z -0.0025 -Y 0.0 -topic robot_description --ros-args -r __ns:=/'].
```

### To Reproduce

I run this command 
ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py

ajjaya18, 2024-03-25:
@hilary-luo  help me please

hilary-luo, 2024-03-25:
@ajjaya18 Please do not create so many issues about the same problem. I will respond when I am able to and it is detrimental to everyone for you to open duplicate tickets. I spend time cleaning up the tickets instead of helping people with their issues. Your issue is a troubleshooting issue and you have already entered a ticket in the appropriate repo for a troubleshooting issue. The "Bug" issue type is only to be used by those who have identified a specific bug in the software. I have responded to you on one of your other tickets, please respond to me there. If that error output on that ticket is out of date then please update it on that ticket.

hilary-luo, 2024-03-25:
For reference for anyone who comes across this ticket, you can view the troubleshooting process here: https://github.com/turtlebot/turtlebot4/issues/383

closed 2024-03-25 by hilary-luo