turtlebot/turtlebot4_simulator #71: Humble + Gazebo Garden + PX4

federicociresola, 2024-03-27:
### Describe the the feature you would like

Hi,
Is it possible to make a porting of the turtlebot4-sim on Gazebo Garden? Is it difficult? I am available to help you to make it
because I need it for a work in my university

### Motivation and impact

I am working with PX4 v1.14 and ti work only with Gazebo Garden or Harmonic

RustyCPR, 2024-06-25:
Hi there @federicociresola 
Thanks for your question. 
Reviewing this internally its not something we are looking to support currently. 
See additional details on the ROS / Gazebo compatibility table: [https://gazebosim.org/docs/garden/ros_installation#summary-of-compatible-ros-and-gazebo-combinations](https://gazebosim.org/docs/garden/ros_installation#summary-of-compatible-ros-and-gazebo-combinations)
Thank you,
Rusty

federicociresola, 2024-06-26:
@RustyCPR  thank you for your answer

closed 2024-06-26 by federicociresola