turtlebot/turtlebot4_simulator #72: Turtlebot 4 Simulator crashes after latest updates

twright, 2024-04-08:
### Robot Model

Select One

### ROS distro

Select One

### Networking Configuration

Select One

### OS

Select One

### Built from source or installed?

Select One

### Package version

1.0.1-1jammy.20240321.134526

### Expected behaviour

Running this command should launch Rviz and the Gazebo simulator

### Actual behaviour

Rviz and Gazebo briefly launch before crashing

### To Reproduce

1. Install ROS humble
2. Install the Turtlebot4 packages via sudo apt-get update && sudo apt-get install -y ros-humble-turtlebot4-desktop ros-humble-turtlebot4-simulator ros-humble-turtlebot4-viz ros-humble-turtlebot4-diagnostics
3. Run ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py  world:=maze slam:=true nav2:=true rviz:=true

### Other notes

I have encountered this bug both running locally on Ubuntu and in a Docker container. I have an Nvidia Geforce RTX 2070 MaxQ dGPU, and this bugs occurs either running on embedded graphics, or launching it on the Nvidia GPU.

twright, 2024-04-08:
The error message is
```
[INFO] [launch]: All log files can be found below /home/au759518/.ros/log/2024-04-08-12-14-04-324353-d30855-3793056
[INFO] [launch]: Default logging verbosity is set to INFO
ign_args is deprecated, migrate to gz_args!
[INFO] [ruby $(which ign) gazebo-1]: process started with pid [3793135]
[INFO] [parameter_bridge-2]: process started with pid [3793137]
[INFO] [robot_state_publisher-3]: process started with pid [3793140]
[INFO] [joint_state_publisher-4]: process started with pid [3793143]
[INFO] [robot_state_publisher-5]: process started with pid [3793145]
[INFO] [static_transform_publisher-6]: process started with pid [3793147]
[INFO] [create-7]: process started with pid [3793149]
[INFO] [create-8]: process started with pid [3793151]
[INFO] [parameter_bridge-9]: process started with pid [3793153]
[INFO] [parameter_bridge-10]: process started with pid [3793156]
[INFO] [parameter_bridge-11]: process started with pid [3793160]
[INFO] [parameter_bridge-12]: process started with pid [3793163]
[INFO] [parameter_bridge-13]: process started with pid [3793165]
[INFO] [parameter_bridge-14]: process started with pid [3793167]
[INFO] [parameter_bridge-15]: process started with pid [3793169]
[INFO] [parameter_bridge-16]: process started with pid [3793171]
[INFO] [parameter_bridge-17]: process started with pid [3793173]
[INFO] [parameter_bridge-18]: process started with pid [3793175]
[INFO] [parameter_bridge-19]: process started with pid [3793177]
[INFO] [parameter_bridge-20]: process started with pid [3793179]
[INFO] [parameter_bridge-21]: process started with pid [3793181]
[INFO] [parameter_bridge-22]: process started with pid [3793183]
[INFO] [parameter_bridge-23]: process started with pid [3793185]
[INFO] [parameter_bridge-24]: process started with pid [3793187]
[INFO] [parameter_bridge-25]: process started with pid [3793189]
[INFO] [parameter_bridge-26]: process started with pid [3793191]
[INFO] [parameter_bridge-27]: process started with pid [3793193]
[INFO] [parameter_bridge-28]: process started with pid [3793195]
[INFO] [parameter_bridge-29]: process started with pid [3793197]
[INFO] [turtlebot4_node-30]: process started with pid [3793211]
[INFO] [turtlebot4_ignition_hmi_node-31]: process started with pid [3793235]
[INFO] [spawner-32]: process started with pid [3793254]
[INFO] [hazards_vector_publisher-33]: process started with pid [3793278]
[INFO] [ir_intensity_vector_publisher-34]: process started with pid [3793331]
[INFO] [motion_control-35]: process started with pid [3793409]
[INFO] [wheel_status_publisher-36]: process started with pid [3793441]
[INFO] [mock_publisher-37]: process started with pid [3793453]
[INFO] [robot_state-38]: process started with pid [3793488]
[INFO] [kidnap_estimator_publisher-39]: process started with pid [3793519]
[INFO] [ui_mgr-40]: process started with pid [3793538]
[INFO] [pose_republisher_node-41]: process started with pid [3793549]
[INFO] [sensors_node-42]: process started with pid [3793552]
[INFO] [interface_buttons_node-43]: process started with pid [3793555]
[INFO] [static_transform_publisher-44]: process started with pid [3793566]
[INFO] [static_transform_publisher-45]: process started with pid [3793569]
[INFO] [sync_slam_toolbox_node-46]: process started with pid [3793571]
[INFO] [controller_server-47]: process started with pid [3793578]
[INFO] [smoother_server-48]: process started with pid [3793581]
[INFO] [planner_server-49]: process started with pid [3793596]
[INFO] [behavior_server-50]: process started with pid [3793602]
[INFO] [bt_navigator-51]: process started with pid [3793619]
[INFO] [waypoint_follower-52]: process started with pid [3793621]
[INFO] [velocity_smoother-53]: process started with pid [3793624]
[INFO] [lifecycle_manager-54]: process started with pid [3793626]
[INFO] [rviz2-55]: process started with pid [3793628]
[parameter_bridge-2] [INFO] [1712571247.142438401] [clock_bridge]: Creating GZ->ROS Bridge: [/clock (ignition.msgs.Clock) -> /clock (rosgraph_msgs/msg/Clock)] (Lazy 0)
[robot_state_publisher-3] [WARN] [1712571247.048395247] [kdl_parser]: The root link base_link has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
[robot_state_publisher-3] [INFO] [1712571247.065356396] [robot_state_publisher]: got segment base_link
[robot_state_publisher-3] [INFO] [1712571247.065427456] [robot_state_publisher]: got segment bump_front_center
[robot_state_publisher-3] [INFO] [1712571247.065439651] [robot_state_publisher]: got segment bump_front_left
[robot_state_publisher-3] [INFO] [1712571247.065445851] [robot_state_publisher]: got segment bump_front_right
[robot_state_publisher-3] [INFO] [1712571247.065451260] [robot_state_publisher]: got segment bump_left
[robot_state_publisher-3] [INFO] [1712571247.065456654] [robot_state_publisher]: got segment bump_right
[robot_state_publisher-3] [INFO] [1712571247.065462707] [robot_state_publisher]: got segment bumper
[robot_state_publisher-3] [INFO] [1712571247.065468948] [robot_state_publisher]: got segment button_1
[... 1159 lines cut ...]
[INFO] [parameter_bridge-11]: process has finished cleanly [pid 3793160]
[parameter_bridge-10] [INFO] [1712571260.370079659] [rclcpp]: signal_handler(signum=2)
[parameter_bridge-9] [INFO] [1712571260.381891641] [rclcpp]: signal_handler(signum=2)
[static_transform_publisher-6] [INFO] [1712571260.394192554] [rclcpp]: signal_handler(signum=2)
[... cut ...]
[ERROR] [rviz2-55]: process has died [pid 3793628, exit code -6, cmd '/opt/ros/humble/lib/rviz2/rviz2 -d /opt/ros/humble/share/turtlebot4_viz/rviz/robot.rviz --ros-args -r __node:=rviz2 -r __ns:=/ --params-file /tmp/launch_params_0r8mj_gg -r /tf:=tf -r /tf_static:=tf_static'].
[ERROR] [controller_server-47]: process has died [pid 3793578, exit code -6, cmd '/opt/ros/humble/lib/nav2_controller/controller_server --ros-args --log-level info --ros-args -r __ns:=/ --params-file /tmp/launch_params_yjnx371d -r /global_costmap/scan:=/scan -r /local_costmap/scan:=/scan -r /tf:=tf -r /tf_static:=tf_static -r cmd_vel:=cmd_vel_nav'].
```

alexdo105, 2024-04-10:
I have the same issue, would appreciate the solution.

hilary-luo, 2024-04-10:
@twright Thanks for bringing this to our attention. This error is from an upstream package and has been fixed. It looks like we will just need to wait until the fix gets pushed. Alternatively you can run the afflicted package from source, see the commit fixing it here: https://github.com/ros-controls/gz_ros2_control/commit/f724da5d37d3a9f88ee38d2aea05899679c3c448

@alexdo105 To confirm that you are experiencing the same issue, can you verify that your output also gives the following error: 
```
[ruby $(which ign) gazebo-1] terminate called after throwing an instance of 'rclcpp::exceptions::ParameterAlreadyDeclaredException'
[ruby $(which ign) gazebo-1]   what():  parameter 'position_proportional_gain' has already been declared
```

It would be in the center of the output so it's best to search for it. If not please open your own ticket so we can troubleshoot the issue you are seeing.

alexdo105, 2024-04-11:
@hilary-luo The error that I am having now is the following:
URL is deprecrated. Pleasse change https://fuel.ignitionrobotics.org to https://fuel.gazebosim.org
[ruby $(which ign) gazebo-1] [Err] [RestClient.cc:343] Error in REST request
[ruby $(which ign) gazebo-1] 
[ruby $(which ign) gazebo-1] libcurl: (6) Could not resolve host: fuel.ignitionrobotics.org
[ruby $(which ign) gazebo-1] [Err] [FuelClient.cc:717] Failed to download model.
[ruby $(which ign) gazebo-1]   Server: https://fuel.ignitionrobotics.org
[ruby $(which ign) gazebo-1]   Route: openrobotics/models/table/tip/table.zip
[ruby $(which ign) gazebo-1]   REST response code: 0
[ruby $(which ign) gazebo-1] [Wrn] [FuelClient.cc:1978] The fuel.ignitionrobotics.org URL is deprecrated. Pleasse change https://fuel.ignitionrobotics.org to https://fuel.gazebosim.org
[ruby $(which ign) gazebo-1] [Err] [RestClient.cc:343] Error in REST request
[ruby $(which ign) gazebo-1] 
[ruby $(which ign) gazebo-1] libcurl: (6) Could not resolve host: fuel.ignitionrobotics.org
[ruby $(which ign) gazebo-1] [Err] [FuelClient.cc:717] Failed to download model.
[ruby $(which ign) gazebo-1]   Server: https://fuel.ignitionrobotics.org
[ruby $(which ign) gazebo-1]   Route: openrobotics/models/artifact proximity detector/tip/artifact proximity detector.zip
[ruby $(which ign) gazebo-1]   REST response code: 0
[ruby $(which ign) gazebo-1] [GUI] [Wrn] [Application.cc:797] [QT] qrc:/qml/StyleDialog.qml:112:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[ruby $(which ign) gazebo-1] [GUI] [Wrn] [Application.cc:797] [QT] qrc:/qml/StyleDialog.qml:105:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[ruby $(which ign) gazebo-1] [GUI] [Wrn] [Application.cc:797] [QT] qrc:/qml/StyleDialog.qml:98:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[ruby $(which ign) gazebo-1] [GUI] [Wrn] [Application.cc:797] [QT] qrc:qml/Main.qml:102:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[ruby $(which ign) gazebo-1] [GUI] [Wrn] [Application.cc:797] [QT] qrc:/qml/PluginMenu.qml:27:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[ruby $(which ign) gazebo-1] [Err] [Server.cc:139] Error Code 13: [/sdf/world[@name="warehouse"]/include[1]/uri:/opt/ros/humble/share/turtlebot4_ignition_bringup/worlds/warehouse.sdf:L54]: Msg: Unable to find uri[https://fuel.ignitionrobotics.org/1.0/MovAi/models/pallet_box_mobile]
[ruby $(which ign) gazebo-1] [Err] [Server.cc:139] Error Code 13: [/sdf/world[@name="warehouse"]/include[2]/uri:/opt/ros/humble/share/turtlebot4_ignition_bringup/worlds/warehouse.sdf:L64]: Msg: Unable to find uri[https://fuel.ignitionrobotics.org/1.0/MovAi/models/shelf_big]
[ruby $(which ign) gazebo-1] [Err] [Server.cc:139] Error Code 13: [/sdf/world[@name="warehouse"]/include[3]/uri:/opt/ros/humble/share/turtlebot4_ignition_bringup/worlds/warehouse.sdf:L71]: Msg: Unable to find uri[https://fuel.ignitionrobotics.org/1.0/MovAi/models/shelf_big]
[ruby $(which ign) gazebo-1] [Err] [Server.cc:139] Error Code 13: [/sdf/world[@name="warehouse"]/include[4]/uri:/opt/ros/humble/share/turtlebot4_ignition_bringup/worlds/warehouse.sdf:L78]: Msg: Unable to find uri[https://fuel.ignitionrobotics.org/1.0/MovAi/models/shelf_big]
[ruby $(which ign) gazebo-1] [Err] [Server.cc:139] Error Code 13: [/sdf/world[@name="warehouse"]/include[5]/uri:/opt/ros/humble/share/turtlebot4_ignition_bringup/worlds/warehouse.sdf:L85]: Msg: Unable to find uri[https://fuel.ignitionrobotics.org/1.0/MovAi/models/shelf]
[ruby $(which ign) gazebo-1] [Err] [Server.cc:139] Error Code 13: [/sdf/world[@name="warehouse"]/include[6]/uri:/opt/ros/humble/share/turtlebot4_ignition_bringup/worlds/warehouse.sdf:L92]: Msg: Unable to find uri[https://fuel.ignitionrobotics.org/1.0/MovAi/models/shelf]
[ruby $(which ign) gazebo-1] [Err] [Server.cc:139] Error Code 13: [/sdf/world[@name="warehouse"]/include[7]/uri:/opt/ros/humble/share/turtlebot4_ignition_bringup/worlds/warehouse.sdf:L99]: Msg: Unable to find uri[https://fuel.ignitionrobotics.org/1.0/MovAi/models/shelf]
[ruby $(which ign) gazebo-1] [Err] [Server.cc:139] Error Code 13: [/sdf/world[@name="warehouse"]/include[8]/uri:/opt/ros/humble/share/turtlebot4_ignition_bringup/worlds/warehouse.sdf:L106]: Msg: Unable to find uri[https://fuel.ignitionrobotics.org/1.0/MovAi/models/shelf]
[ruby $(which ign) gazebo-1] [Err] [Server.cc:139] Error Code 13: [/sdf/world[@name="warehouse"]/include[9]/uri:/opt/ros/humble/share/turtlebot4_ignition_bringup/worlds/warehouse.sdf:L113]: Msg: Unable to find uri[https://fuel.ignitionrobotics.org/1.0/MovAi/models/shelf]
[ruby $(which ign) gazebo-1] [Err] [Server.cc:139] Error Code 13: [/sdf/world[@name="warehouse"]/include[10]/uri:/opt/ros/humble/share/turtlebot4_ignition_bringup/worlds/warehouse.sdf:L123]: Msg: Unable to find uri[https://fuel.ignitionrobotics.org/1.0/MovAi/models/shelf_big]
[... cut ...]
[ruby $(which ign) gazebo-1] [GUI] [Dbg] [PathManager.cc:66] Requesting resource paths through [/gazebo/resource_paths/get]
[ruby $(which ign) gazebo-1] [GUI] [Dbg] [Gui.cc:333] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[ruby $(which ign) gazebo-1] [GUI] [Dbg] [SignalHandler.cc:141] Received signal[2].
[ruby $(which ign) gazebo-1] [GUI] [Dbg] [SignalHandler.cc:141] Received signal[2].
[ruby $(which ign) gazebo-1] [GUI] [Dbg] [Application.cc:140] Terminating application.

hilary-luo, 2024-04-11:
@alexdo105 That is addressed here: https://github.com/turtlebot/turtlebot4_simulator/pull/70

Either you can run it from source or wait for the update to come through. Alternatively you can manually download the files as mentioned here: https://github.com/clearpathrobotics/clearpath_simulator/issues/30

ajjaya18, 2024-04-21:
Hi, Hello, when will this problem be solved? Because I also try with maze and with warehouse, and I am having the same issue. It is of utmost importance that it works for me because it would help me a lot since I am working on this

ajjaya18, 2024-04-21:
If I launch my own world, will the problem be solved or not?

closed 2024-04-30 by hilary-luo

alexdo105, 2024-05-02:
> The updated package was synced. Please update your packages and this issue
> should be resolved for everyone.

Hello Hillary, thanks for updating us. One question, has a change been made
to the way that the initial pose is set? Before this update I called
navigator.setInitialPose to set it and it worked, but after this update
something seems to have changed, beginning with that now there is a topic
called “initialpose”…
Thanks again!

hilary-luo, 2024-05-03:
@alexdo105 all of the navigation is handled by the nav2 packages which the Turtlebot4 packages launch. If something is nonfunctional now in the Turtlebot4 packages / tutorials then please make a ticket including the full terminal output. If you are just looking for information about the latest updates in nav2 I would suggest looking at the recent commits on the [nav2 repo](https://github.com/open-navigation/docs.nav2.org).