turtlebot/turtlebot4_simulator #73: Depot world fails to download / won't load

hilary-luo, 2024-04-11:
### Robot Model

Select One

### ROS distro

Select One

### Networking Configuration

Select One

### OS

Select One

### Built from source or installed?

Select One

### Package version

All

### Expected behaviour

When you try to launch the depot world for the first time, it should download.

### Actual behaviour

When you try to launch the depot world for the first time, it is not currently able to be downloaded. This is caused by this issue: https://github.com/gazebo-web/fuel-server/issues/19

### Error messages

```bash
[ruby $(which ign) gazebo-1] [Err] [FuelClient.cc:717] Failed to download model.
[ruby $(which ign) gazebo-1]   Server: https://fuel.ignitionrobotics.org/
```

### To Reproduce

Install simulation as per instructions and try to run the depot world simulation.

Kavya-812, 2024-04-12:
Facing the same issue. Please give the resolve. Already tried manually updating the links.

Kavya-812, 2024-04-12:
ros2 launch turtlebot4_ignition_bringup ignition.launch.py
[INFO] [launch]: All log files can be found below /home/kavya/.ros/log/2024-04-12-18-56-49-363474-kavya-Victus-by-HP-Gaming-Laptop-15-fa0xxx-4268
[INFO] [launch]: Default logging verbosity is set to INFO
ign_args is deprecated, migrate to gz_args!
[INFO] [ruby $(which ign) gazebo-1]: process started with pid [4269]
[INFO] [parameter_bridge-2]: process started with pid [4272]
[parameter_bridge-2] [INFO] [1712928409.422855394] [clock_bridge]: Creating GZ->ROS Bridge: [/clock (ignition.msgs.Clock) -> /clock (rosgraph_msgs/msg/Clock)] (Lazy 0)
[ruby $(which ign) gazebo-1] Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
[ruby $(which ign) gazebo-1] [Wrn] [FuelClient.cc:1978] The fuel.ignitionrobotics.org URL is deprecrated. Pleasse change https://fuel.ignitionrobotics.org to https://fuel.gazebosim.org
[ruby $(which ign) gazebo-1] [Err] [RestClient.cc:343] Error in REST request
[ruby $(which ign) gazebo-1] 
[ruby $(which ign) gazebo-1] libcurl: (6) Could not resolve host: fuel.ignitionrobotics.org
[ruby $(which ign) gazebo-1] [Err] [FuelClient.cc:717] Failed to download model.
[ruby $(which ign) gazebo-1]   Server: https://fuel.ignitionrobotics.org
[ruby $(which ign) gazebo-1]   Route: openrobotics/models/warehouse/tip/warehouse.zip
[ruby $(which ign) gazebo-1]   REST response code: 0
[ruby $(which ign) gazebo-1] [Wrn] [FuelClient.cc:1978] The fuel.ignitionrobotics.org URL is deprecrated. Pleasse change https://fuel.ignitionrobotics.org to https://fuel.gazebosim.org
[ruby $(which ign) gazebo-1] [Err] [RestClient.cc:343] Error in REST request
[ruby $(which ign) gazebo-1] 
[ruby $(which ign) gazebo-1] libcurl: (6) Could not resolve host: fuel.ignitionrobotics.org
[ruby $(which ign) gazebo-1] [Err] [FuelClient.cc:717] Failed to download model.
[ruby $(which ign) gazebo-1]   Server: https://fuel.ignitionrobotics.org
[ruby $(which ign) gazebo-1]   Route: movai/models/pallet_box_mobile/tip/pallet_box_mobile.zip
[ruby $(which ign) gazebo-1]   REST response code: 0
[ruby $(which ign) gazebo-1] [Wrn] [FuelClient.cc:1978] The fuel.ignitionrobotics.org URL is deprecrated. Pleasse change https://fuel.ignitionrobotics.org to https://fuel.gazebosim.org
[ruby $(which ign) gazebo-1] [Err] [RestClient.cc:343] Error in REST request
[ruby $(which ign) gazebo-1] 
[ruby $(which ign) gazebo-1] libcurl: (6) Could not resolve host: fuel.ignitionrobotics.org
[ruby $(which ign) gazebo-1] [Err] [FuelClient.cc:717] Failed to download model.
[ruby $(which ign) gazebo-1]   Server: https://fuel.ignitionrobotics.org
[ruby $(which ign) gazebo-1]   Route: movai/models/shelf_big/tip/shelf_big.zip
[ruby $(which ign) gazebo-1]   REST response code: 0
[ruby $(which ign) gazebo-1] [Wrn] [FuelClient.cc:1978] The fuel.ignitionrobotics.org URL is deprecrated. Pleasse change https://fuel.ignitionrobotics.org to https://fuel.gazebosim.org
[ruby $(which ign) gazebo-1] [Err] [RestClient.cc:343] Error in REST request
[ruby $(which ign) gazebo-1] 
[ruby $(which ign) gazebo-1] libcurl: (6) Could not resolve host: fuel.ignitionrobotics.org
[ruby $(which ign) gazebo-1] [Err] [FuelClient.cc:717] Failed to download model.
[ruby $(which ign) gazebo-1]   Server: https://fuel.ignitionrobotics.org
[ruby $(which ign) gazebo-1]   Route: movai/models/shelf_big/tip/shelf_big.zip
[ruby $(which ign) gazebo-1]   REST response code: 0
[ruby $(which ign) gazebo-1] [Wrn] [FuelClient.cc:1978] The fuel.ignitionrobotics.org URL is deprecrated. Pleasse change https://fuel.ignitionrobotics.org to https://fuel.gazebosim.org
[ruby $(which ign) gazebo-1] [GUI] [Wrn] [Application.cc:797] [QT] qrc:/qml/StyleDialog.qml:112:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[ruby $(which ign) gazebo-1] [GUI] [Wrn] [Application.cc:797] [QT] qrc:/qml/StyleDialog.qml:105:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[ruby $(which ign) gazebo-1] [GUI] [Wrn] [Application.cc:797] [QT] qrc:/qml/StyleDialog.qml:98:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[ruby $(which ign) gazebo-1] [GUI] [Wrn] [Application.cc:797] [QT] qrc:qml/Main.qml:102:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[ruby $(which ign) gazebo-1] [GUI] [Wrn] [Application.cc:797] [QT] qrc:/qml/PluginMenu.qml:27:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[ruby $(which ign) gazebo-1] [Err] [RestClient.cc:343] Error in REST request
[ruby $(which ign) gazebo-1] 
[ruby $(which ign) gazebo-1] libcurl: (6) Could not resolve host: fuel.ignitionrobotics.org
[ruby $(which ign) gazebo-1] [Err] [FuelClient.cc:717] Failed to download model.
[ruby $(which ign) gazebo-1]   Server: https://fuel.ignitionrobotics.org
[ruby $(which ign) gazebo-1]   Route: movai/models/shelf_big/tip/shelf_big.zip
[ruby $(which ign) gazebo-1]   REST response code: 0
[... cut ...]
[ruby $(which ign) gazebo-1] [GUI] [Dbg] [Application.cc:140] Terminating application.
[INFO] [ruby $(which ign) gazebo-1]: process has finished cleanly [pid 4269]
[INFO] [launch]: process[ruby $(which ign) gazebo-1] was required: shutting down launched system
[INFO] [parameter_bridge-2]: sending signal 'SIGINT' to process[parameter_bridge-2]
[parameter_bridge-2] [INFO] [1712928414.873383001] [rclcpp]: signal_handler(signum=2)
[INFO] [parameter_bridge-2]: process has finished cleanly [pid 4272]

Kavya-812, 2024-04-12:
Also when running in the maze  world the bot is not spawning

ros2 launch turtlebot4_ignition_bringup ignition.launch.py world:=maze model:=lite

[INFO] [launch]: All log files can be found below /home/kavya/.ros/log/2024-04-12-19-21-40-431215-kavya-Victus-by-HP-Gaming-Laptop-15-fa0xxx-16547
[INFO] [launch]: Default logging verbosity is set to INFO
ign_args is deprecated, migrate to gz_args!
[INFO] [ruby $(which ign) gazebo-1]: process started with pid [16548]
[INFO] [parameter_bridge-2]: process started with pid [16551]
[parameter_bridge-2] [INFO] [1712929900.491571569] [clock_bridge]: Creating GZ->ROS Bridge: [/clock (ignition.msgs.Clock) -> /clock (rosgraph_msgs/msg/Clock)] (Lazy 0)
[ruby $(which ign) gazebo-1] Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
[ruby $(which ign) gazebo-1] [Dbg] [gz.cc:161] Subscribing to [/gazebo/starting_world].
[ruby $(which ign) gazebo-1] [Dbg] [gz.cc:163] Waiting for a world to be set from the GUI...
[ruby $(which ign) gazebo-1] [Msg] Received world [maze.sdf] from the GUI.
[ruby $(which ign) gazebo-1] [Dbg] [gz.cc:167] Unsubscribing from [/gazebo/starting_world].
[ruby $(which ign) gazebo-1] [Msg] Ignition Gazebo Server v6.16.0
[ruby $(which ign) gazebo-1] [Msg] Loading SDF world file[/opt/ros/humble/share/turtlebot4_ignition_bringup/worlds/maze.sdf].
[ruby $(which ign) gazebo-1] [Msg] Serving entity system service on [/entity/system/add]
[ruby $(which ign) gazebo-1] [Dbg] [Physics.cc:804] Loaded [ignition::physics::dartsim::Plugin] from library [/usr/lib/x86_64-linux-gnu/ign-physics-5/engine-plugins/libignition-physics-dartsim-plugin.so]
[ruby $(which ign) gazebo-1] [Dbg] [SystemManager.cc:70] Loaded system [ignition::gazebo::systems::Physics] for entity [1]
[ruby $(which ign) gazebo-1] [Msg] Create service on [/world/maze/create]
[ruby $(which ign) gazebo-1] [Msg] Remove service on [/world/maze/remove]
[ruby $(which ign) gazebo-1] [Msg] Pose service on [/world/maze/set_pose]
[ruby $(which ign) gazebo-1] [Msg] Pose service on [/world/maze/set_pose_vector]
[ruby $(which ign) gazebo-1] [Msg] Light configuration service on [/world/maze/light_config]
[ruby $(which ign) gazebo-1] [Msg] Physics service on [/world/maze/set_physics]
[ruby $(which ign) gazebo-1] [Msg] SphericalCoordinates service on [/world/maze/set_spherical_coordinates]
[ruby $(which ign) gazebo-1] [Msg] Enable collision service on [/world/maze/enable_collision]
[ruby $(which ign) gazebo-1] [Msg] Disable collision service on [/world/maze/disable_collision]
[ruby $(which ign) gazebo-1] [Msg] Material service on [/world/maze/visual_config]
[ruby $(which ign) gazebo-1] [Msg] Material service on [/world/maze/wheel_slip]
[ruby $(which ign) gazebo-1] [Dbg] [SystemManager.cc:70] Loaded system [ignition::gazebo::systems::UserCommands] for entity [1]
[ruby $(which ign) gazebo-1] [Dbg] [SystemManager.cc:70] Loaded system [ignition::gazebo::systems::SceneBroadcaster] for entity [1]
[ruby $(which ign) gazebo-1] [Dbg] [SystemManager.cc:70] Loaded system [ignition::gazebo::systems::Contact] for entity [1]
[ruby $(which ign) gazebo-1] [Msg] Loaded level [3]
[ruby $(which ign) gazebo-1] [Msg] Serving world controls on [/world/maze/control], [/world/maze/control/state] and [/world/maze/playback/control]
[ruby $(which ign) gazebo-1] [Msg] Serving GUI information on [/world/maze/gui/info]
[ruby $(which ign) gazebo-1] [Msg] World [maze] initialized with [1ms] physics profile.
[ruby $(which ign) gazebo-1] [Msg] Serving world SDF generation service on [/world/maze/generate_world_sdf]
[ruby $(which ign) gazebo-1] [Msg] Serving world names on [/gazebo/worlds]
[ruby $(which ign) gazebo-1] [Msg] Resource path add service on [/gazebo/resource_paths/add].
[ruby $(which ign) gazebo-1] [Msg] Resource path get service on [/gazebo/resource_paths/get].
[ruby $(which ign) gazebo-1] [GUI] [Wrn] [Application.cc:797] [QT] qrc:/qml/StyleDialog.qml:112:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[ruby $(which ign) gazebo-1] [GUI] [Wrn] [Application.cc:797] [QT] qrc:/qml/StyleDialog.qml:105:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[ruby $(which ign) gazebo-1] [GUI] [Wrn] [Application.cc:797] [QT] qrc:/qml/StyleDialog.qml:98:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[ruby $(which ign) gazebo-1] [GUI] [Wrn] [Application.cc:797] [QT] qrc:qml/Main.qml:102:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[ruby $(which ign) gazebo-1] [GUI] [Wrn] [Application.cc:797] [QT] qrc:/qml/PluginMenu.qml:27:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[ruby $(which ign) gazebo-1] [GUI] [Wrn] [Application.cc:797] [QT] file::/Gazebo/GazeboDrawer.qml:242:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[ruby $(which ign) gazebo-1] [GUI] [Wrn] [Scene3D.cc:2941] The GzScene3D plugin is deprecated on v6 and will be removed on v7. Use MinimalScene together with other plugins as needed.
[... cut ...]
[ruby $(which ign) gazebo-1] [GUI] [Wrn] [Model.hh:98] Unable to deserialize sdf::Model
[ruby $(which ign) gazebo-1] [GUI] [Wrn] [Model.hh:98] Unable to deserialize sdf::Model
[ruby $(which ign) gazebo-1] [GUI] [Wrn] [Model.hh:98] Unable to deserialize sdf::Model
[ruby $(which ign) gazebo-1] [GUI] [Wrn] [Model.hh:98] Unable to deserialize sdf::Model
[ruby $(which ign) gazebo-1] [GUI] [Wrn] [Model.hh:98] Unable to deserialize sdf::Model
[ruby $(which ign) gazebo-1] [GUI] [Wrn] [Model.hh:98] Unable to deserialize sdf::Model

JHPHELAN, 2024-04-13:
Same problem.  Uninstalled & reinstalled from source today.
Starts to load, brings up Gazebo window, then window closes and program crashes.
Sometimes window will persist but no Gazebo, just leftover text if moved to another monitor, then fruitless request for world names.  controller_manager spawn seems to be a  problem.  
Script error file attached.
Will fixes apply to Debian package, too, or just source?
[TurtleBot4Errors2024.04.13.txt](https://github.com/turtlebot/turtlebot4_simulator/files/14968368/TurtleBot4Errors2024.04.13.txt)

hilary-luo, 2024-04-14:
@Kavya-812 @JHPHELAN This issue is only about the depot and the current resolution for that world is stated in the ticket: 
> Do not use the depot world until this is resolved. If you really need to then you have to manually download the file from the fuel server website and place it in the correct location.

Any discussion about how this issue is being handled in the other worlds has already been clarified in other tickets and if you feel like you have a question that is not covered, then follow up on those tickets, not this one. This ticket will remain only about the depot world.

@Kavya-812 in regards to your warehouse logs see https://github.com/turtlebot/turtlebot4_simulator/issues/72#issuecomment-2050479333 or there are several others where we have discussed this issue already. In general in the future, if you have an error output that does not match a given ticket then do not put your terminal output as a response to that ticket. Please create your own troubleshooting ticket if you can't tell if your issue matches an existing issue. In regards to your Maze output, I would suggest upgrading to Humble.

@JHPHELAN Please go back to my [response to you on Thursdasy](https://github.com/turtlebot/turtlebot4_simulator/issues/59#issuecomment-2050231327) and follow the requested steps. I asked you to verify your error output against the two links there and if your issue did not match those then to open your own troubleshooting issue. We will not be doing a deep dive into your unique issues in the comments on other peoples' tickets. I do not know which fixes you are referring to as there are no fixes mentioned on this ticket, only a current issue with the depot world. What I can say is that all humble fixes will be released into debian but we do not have full control over when they are made available over apt.

closed 2024-08-26 by hilary-luo