turtlebot/turtlebot4_simulator #78: Jazzy Jalisco Update

lekoook, 2024-06-04:
### Describe the the feature you would like

Update to integrate with ROS2 Jazzy Jalisco release.

### Motivation and impact

This will allow developers using the Jazzy release to continue using the simulation packages.

dragonblade316, 2024-10-21:
The branch appears to be created. 

One thing I noticed is that while the docs say that the Jazzy release is installable, but packages.ros.org does not appear to hold the meta packages. If we are supposed to use a different package repo for the turtlebot sim, I cannot find it in the docs.

Can someone tell me if this is something that needs fixed or if I just missed something obvious.

On a slightly unrelated note, since Jazzy is now in the docs, the issue templates should probably be updated to include Jazzy and Ubuntu 24.04 Noble options.

Just things I noticed while trying to get this working on Jazzy. I will probably just build from source until the package in released.

civerachb-cpr, 2024-10-21:
It looks like one of the dependencies for the Create3 is failing to build, and that's holding up the the debian of the Turtlebot4 simulator package: https://build.ros2.org/job/Jbin_uN64__irobot_create_nodes__ubuntu_noble_amd64__binary

I've reached out to our colleagues at iRobot to inform them of the problem. Hopefully they'll be able to resolve the bug with the dependency, which should unblock the rest of the packages building.

dragonblade316, 2024-10-26:
It looks like the package was just successfully built.

Is there a timeline for when a package is successfully built and uploaded to the repo? Or am I getting overhyped and there are still package problems?

(I don't want to seem impatient, but I just noticed that the package appeared to be fixed and am curious as to what the process looks like from here)

[]( https://build.ros2.org/job/Jbin_uN64__irobot_create_nodes__ubuntu_noble_amd64__binary  )

civerachb-cpr, 2024-10-28:
Once the package is built it's available on the OSRF's testing server. You can enable that by editing your `/etc/apt/sources.list.d/ros2.list` to use the testing server:
```
deb [signed-by=/usr/share/keyrings/ros-archive-keyring.gpg] http://packages.ros.org/ros2-testing/ubuntu noble main
```
Then run `sudo apt-get update` and you should have access to everything on the testing server, including the newly-built packages.

Otherwise, you'll need to wait for the next Jazzy sync. These are announced on discourse.ros.org ahead of time. They basically schedule a bulk copy of the testing packages over to the main apt package server at regular intervals, just as a way to manage the sheer number of packages. I don't know when the next Jazzy sync will be, but they usually do at least one per month.