turtlebot/turtlebot4_simulator #85: I am facing the same issue, but more weird that only a semi-circle is seen when we use the Visualise LIDAR plugin.

harishkumarbalaji, 2024-10-23:
I am facing the same issue, but more weird that only a semi-circle is seen when we use the Visualise LIDAR plugin.
[image]

Nvidia-smi output

```hkb@hkb:~/turtlebot4_ws$ nvidia-smi
Wed Oct 23 17:41:59 2024       
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 550.107.02             Driver Version: 550.107.02     CUDA Version: 12.4     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA GeForce GTX 1660 ...    Off |   00000000:01:00.0 Off |                  N/A |
| N/A   53C    P0             18W /   60W |       6MiB /   6144MiB |      0%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
                                                                                         
+-----------------------------------------------------------------------------------------+
| Processes:                                                                              |
|  GPU   GI   CI        PID   Type   Process name                              GPU Memory |
|        ID   ID                                                               Usage      |
|=========================================================================================|
|    0   N/A  N/A      2058      G   /usr/lib/xorg/Xorg                              4MiB |
+-----------------------------------------------------------------------------------------+```

_Originally posted by @harishkumarbalaji in https://github.com/turtlebot/turtlebot4_simulator/issues/54#issuecomment-2433685168_

aaronwilhelm1, 2024-10-25:
Have you tried changing the render_engine to Ogre2 in irobot_create_description/urdf/create3.urdf.xacro as mentioned in [this](https://github.com/turtlebot/turtlebot4_simulator/issues/20#issuecomment-1302831141) thread? I had a similar problem and that fixed it for me.

harishkumarbalaji, 2024-10-25:
> Have you tried changing the render_engine to Ogre2 in irobot_create_description/urdf/create3.urdf.xacro as mentioned in [this](https://github.com/turtlebot/turtlebot4_simulator/issues/20#issuecomment-1302831141) thread? I had a similar problem and that fixed it for me.

Thanks for the reply. After upgrading it, I have the following observations,
- The Lidar rays now have a proper range, but only semi-circle(or half) rays are still seen, not in all directions.
- The RTF reduces below 1% and it is very laggy.
- My Nvidia Graphics card is not being used and it is in 0% utilization

Any idea on how to make ign gazebo run on a Graphics card instead of a CPU?

aaronwilhelm1, 2024-10-25:
Hmm, I'm not sure what's going on with the semi-circle, unfortunately it's beyond my knowledge. I thought that by default it should run on the GPU, I'm thinking that there might be a setup problem with drivers or incompatibility or a setting you need to change. Sorry I can't help you further

hilary-luo, 2024-10-29:
@harishkumarbalaji Try changing your Nvidia profile settings and setting the GPU to always be used (`PRIME Profiles` -> `NVIDIA (Performance Mode)`).
[image]

harishkumarbalaji, 2024-10-30:
Thanks for the suggestion, @hilary-luo after changing this, the gazebo is running on Nvidia Graphics now, the ray's range is normal, and FPS has not dropped more, but still the semi-circle problem is seen.

[image]

hilary-luo, 2024-11-06:
@harishkumarbalaji Is this issue persistent for you? Can you share information about your setup? are you running native Ubuntu, latest packages? Is this Humble or Jazzy? I haven't found any potential causes for this at this point.

harishkumarbalaji, 2024-11-07:
> @harishkumarbalaji Is this issue persistent for you? Can you share information about your setup? are you running native Ubuntu, latest packages? Is this Humble or Jazzy? I haven't found any potential causes for this at this point.

Yes, this issue seems to be persistent. Currently, when I use Visualize Lidar, I see the following output.
[image]

Following are the setup information,
- Running natively on humble with Ubuntu 22.04.5 LTS
- RAM - 16GB
- CPU - i7 9th Gen
- VRAM - GTX 1660 Ti 6GB 
- Latest commit (running source build of `origin/humble` instead of prebuilt binaries for both Create 3 and Turtlebot 4 repos)
- output of `colcon list -n` in my workspace
 ```
irobot_create_common_bringup
irobot_create_control
irobot_create_description
irobot_create_gazebo_bringup
irobot_create_gazebo_plugins
irobot_create_gazebo_sim
irobot_create_ignition_bringup
irobot_create_ignition_plugins
irobot_create_ignition_sim
irobot_create_ignition_toolbox
irobot_create_msgs
irobot_create_nodes
irobot_create_toolbox
turtlebot4_description
turtlebot4_desktop
turtlebot4_ignition_bringup
turtlebot4_ignition_gui_plugins
turtlebot4_ignition_toolbox
turtlebot4_msgs
turtlebot4_navigation
turtlebot4_node
turtlebot4_simulator
turtlebot4_viz
```

hilary-luo, 2024-11-25:
@harishkumarbalaji Unfortunately I haven't had much time to really look into this further and I haven't found any other samples of this happening. The only thing I can think of at this point to test is: do you still have this issue if you run the sim using binaries?

hilary-luo, 2025-01-13:
@harishkumarbalaji Please update your turtlebot4 to the latest change on the Turtlebot4 repo. I didn't realize that a change had been made that caused only the left side of the lidar to show up. That has now been reverted and should be back to normal operation. Please let me know if there are any issues still outstanding.

harishkumarbalaji, 2025-01-13:
@hilary-luo Thanks for the update. It is fixed now. Even I overlooked those parameters.

closed 2025-01-13 by harishkumarbalaji