turtlebot/turtlebot4_simulator #89: Dynamic Discovery Range Issue

CurtinRobotAI, 2025-02-17:
### Understand?

Yes

CurtinRobotAI, 2025-02-17:
I am using TB4 simulator on ROS2 Jazzy and Gazebo Harmonic. 

We have many users within the same network to practice simulations and each of them should NOT interference each other.  Hence, we set  "ROS_AUTOMATIC_DISCOVERY_RANGE=LOCALHOST" for all desktops. However, these topics are not available for all other terminals within the same desktop running the simulation.  We do not have this issue if set "ROS_AUTOMATIC_DISCOVERY_RANGE=SUBNET", but all messages and topics will be broadcasted to other desktops and cause chaos for all users within the same network.

hilary-luo, 2025-02-18:
@CurtinRobotAI You did not follow the instructions in the template. I am closing this issue, please open a troubleshooting issue in the main [turtlebot4](https://github.com/turtlebot/turtlebot4/issues) repo for troubleshooting assistance.

closed 2025-02-18 by hilary-luo