turtlebot/turtlebot4_simulator #92: Wrong lidar readings

paoloelle, 2025-05-17:
### Robot Model

Turtlebot4 Standard

### ROS distro

Jazzy

### Networking Configuration

I do not know

### OS

Ubuntu 24.04

### Built from source or installed?

Built from Source

### Package version

I used a version from my personal Git [repository](https://github.com/paoloelle/turtlebot4_simulator/tree/jazzy). Anyway, my coolleagues that installed it from debian package, by following the instruction in the [User manual](https://github.com/paoloelle/turtlebot4_simulator/tree/jazzy) have the same problem.

### Expected behaviour

The lidar readings should be the correct value.

### Actual behaviour

The lidar readings from the `/scan` topic are always the same (wrong) value. 

### Error messages

```bash
header:
  stamp:
    sec: 1
    nanosec: 491000000
  frame_id: turtlebot4/rplidar_link/rplidar
angle_min: -3.1415927410125732
angle_max: 3.1415927410125732
angle_increment: 0.009832840412855148
time_increment: 0.0
scan_time: 0.0
range_min: 0.164000004529953
range_max: 12.0
ranges:
- 0.164000004529953
- 0.164000004529953
- 0.164000004529953
- 0.164000004529953
- 0.164000004529953
- 0.164000004529953
- 0.164000004529953
- 0.164000004529953
- 0.164000004529953
- 0.164000004529953
- 0.164000004529953
- 0.164000004529953
- 0.164000004529953
- 0.164000004529953
- 0.164000004529953
- 0.164000004529953
- 0.164000004529953
- 0.164000004529953
- 0.164000004529953
- 0.164000004529953
- 0.164000004529953
- 0.164000004529953
[... 587 lines cut ...]
[gazebo-1] [GUI] [Msg] Listening to stats on [/world/warehouse/stats]
[gazebo-1] [GUI] [Msg] Added plugin [World stats] to main window
[gazebo-1] [GUI] [Msg] Loaded plugin [WorldStats] from path [/opt/ros/jazzy/opt/gz_gui_vendor/lib/gz-gui-8/plugins/libWorldStats.so]
[gazebo-1] [GUI] [Dbg] [Application.cc:398] Loading window config
[gazebo-1] [GUI] [Msg] Using server control service [/server_control]
[gazebo-1] [GUI] [Dbg] [Application.cc:671] Applying config
[gazebo-1] [GUI] [Dbg] [MinimalScene.cc:802] Creating gz-rendering interface for OpenGL
[gazebo-1] [GUI] [Dbg] [MinimalScene.cc:986] Creating render thread interface for OpenGL
[gazebo-1] [GUI] [Msg] Loading plugin [gz-rendering-ogre2]
[gazebo-1] [GUI] [Dbg] [MinimalScene.cc:749] Create scene [scene]
[gazebo-1] [GUI] [Dbg] [MinimalScene.cc:1037] Creating texture node render interface for OpenGL
[gazebo-1] [GUI] [Dbg] [InteractiveViewControl.cc:176] InteractiveViewControl plugin is moving camera [scene::Camera(65527)]
[gazebo-1] Warning [Utils.cc:132] [/sdf/model[@name="turtlebot4"]/joint[@name="wheel_drop_left_joint"]/physics/ode/provide_feedback:<data-string>:L1931]: XML Element[provide_feedback], child of element[ode], not defined in SDF. Copying[provide_feedback] as children of [ode].
[gazebo-1] Warning [Utils.cc:132] [/sdf/model[@name="turtlebot4"]/joint[@name="wheel_drop_right_joint"]/physics/ode/provide_feedback:<data-string>:L2052]: XML Element[provide_feedback], child of element[ode], not defined in SDF. Copying[provide_feedback] as children of [ode].
[gazebo-1] [GUI] [Wrn] [Component.hh:172] Trying to deserialize component with data type [St10shared_ptrIN3sdf3v147ElementEE], which doesn't have `operator>>`. Component will not be deserialized.
[gazebo-1] [GUI] [Wrn] [OBJLoader.cc:91] Both `d` and `Tr` parameters defined for "Chair". Use the value of `d` for dissolve (line 8 in .mtl.)
[gazebo-1] 
[gazebo-1] [GUI] [Wrn] [OBJLoader.cc:91] Both `d` and `Tr` parameters defined for "CoffeeTable". Use the value of `d` for dissolve (line 8 in .mtl.)
[gazebo-1] 
[gazebo-1] [GUI] [Dbg] [SignalHandler.cc:142] Received signal[2].
[gazebo-1] [GUI] [Dbg] [Gui.cc:535] Shutting down gz-sim-gui
[gazebo-1] [GUI] [Dbg] [Application.cc:237] Terminating application.
[gazebo-1] [GUI] [Dbg] [MinimalScene.cc:841] Destroy scene [scene]
[INFO] [gazebo-1]: process has finished cleanly [pid 27890]
```

paoloelle, 2025-05-17:
Is it possible that specifying the flag `' --render-engine ogre2'` in the launch file `sim.launch.py` fixed the problem?

closed 2025-06-10 by paoloelle

HustRoboconClub, 2025-06-12:
Thank you very much. Could you explain how to "specifying the flag ' --render-engine ogre2' in the launch file sim.launch.py".
Approach 1, export LIBGL_ALWAYS_SOFTWARE=1 work but slow down simulation
Approach 2, I do not have nvidia GPU, this is my GPU
[image]